microsoft/Windows-driver-samples #2: Enable WPP Recorder does not work

akd-jls, 2015-04-17:
I tried to enable "Enable WPP Recorder" in Sensors Code. The code throws errors. 

Navigate to Properties->Wpp Tracing->Function and Macro Options and choose Enable WPP Recorder.

david1234321, 2015-04-17:
What sample were you hitting this on?

akd-jls, 2015-04-20:
I tried Pedometer driver with VS2013, SDK & WDK File Version 10.0.9926.

https://github.com/Microsoft/Windows-driver-samples/tree/master/sensors/Pedometer

jmrossy, 2015-05-11:
We've reproed this issue and will address it a future kits release. Thanks for your patience.

jmrossy, 2015-05-11:
If this issue is blocking you, there's a workaround you can use. Add the following two lines near the bottom of the SensorsTrace.h header.

    #define WPP_RECORDER_LEVEL_FLAGS_SENSOREXIT_FILTER(LEVEL, FLAGS, status)   WPP_RECORDER_LEVEL_FLAGS_FILTER(LEVEL, FLAGS)
    #define WPP_RECORDER_LEVEL_FLAGS_SENSOREXIT_ARGS(LEVEL, FLAGS, status)     WPP_RECORDER_LEVEL_FLAGS_ARGS(LEVEL, FLAGS)

You mentioned above you're using VS2013. For Windows 10 driver development we strongly recommend using the VS2015 preview with the [latest SDK & WDK releases](https://msdn.microsoft.com/en-US/windows/hardware/dn913721(v=vs8.5).aspx "Windows Insiders").

commit 2e757e5e (2015-07-29): Samples update for public Windows 10 release
Fix #2 Enabling WPP Recorder in Sensors Samples causes errors
Fix #4 Add back fixed KMDOD sample
Add new BarcodeScanner sample in pos folder
Add new MagneticStripeReader sample in pos folder
Add new SynpaticsTouch sample in input folder
Add new Power Engine Plugin sample in pofx folder
Add new DeviceMft sample in avstream folder
Add new AvsCamera sample in root
Add new SimBatt sample in root
Add other pre-existing samples not yet released for Win10
commit 5d61a4a7 (2015-07-29): Samples update for public Windows 10 release
Fix #2 Enabling WPP Recorder in Sensors Samples causes errors
Fix #4 Add back fixed KMDOD sample
Add new BarcodeScanner sample in pos folder
Add new MagneticStripeReader sample in pos folder
Add new SynpaticsTouch sample in input folder
Add new Power Engine Plugin sample in pofx folder
Add new DeviceMft sample in avstream folder
Add new AvsCamera sample in root
Add new SimBatt sample in root
Add other pre-existing samples not yet released for Win10
closed 2015-07-29 by jmrossy