microsoft/Windows-driver-samples #3: NDIS version macro in netvmini

maxtruxa, 2015-04-27:
The [netvmini](https://github.com/Microsoft/Windows-driver-samples/blob/master/network/ndis/netvmini) example uses custom macros ([`MP_NDIS_MAJOR_VERSION`/`MP_NDIS_MINOR_VERSION`](https://github.com/Microsoft/Windows-driver-samples/blob/master/network/ndis/netvmini/6x/miniport.h#L47), written to `NDIS_MINIPORT_DRIVER_CHARACTERISTICS` [here](https://github.com/Microsoft/Windows-driver-samples/blob/master/network/ndis/netvmini/6x/miniport.c#L137)) to specify the used major/minor NDIS version instead of using the macros defined by NDIS.h itself (`NDIS_MINIPORT_MAJOR_VERSION`, `NDIS_MINIPORT_MINOR_VERSION`).
Is there any particular reason to do so?

jtippet, 2015-07-24:
There is not a particular reason to do so.  I suspect that the person who wrote that sample was unaware at the time of `NDIS_MINIPORT_MAJOR_VERSION`.  (Actually I would level all sorts of accusations at the competency of the engineer who wrote that sample, because he was I 😉.)  Anyway, I encourage you to go ahead and use `NDIS_MINIPORT_MAJOR_VERSION` directly.  It eliminates a couple extra lines of code, and makes your driver that much easier to port to a new NDIS version.

jmrossy, 2015-08-02:
Max, do you need additional clarification on this topic? Otherwise I'll close out this issue in a few days.

maxtruxa, 2015-08-03:
Many thanks to Jeffrey for his answer. I'm all set, but wouldn't it be a good idea to add a comment to the documentation of `NDIS_MINIPORT_DRIVER_CHARACTERISTICS` regarding this (e.g. "MajorNdisVersion: This field is typically set to `NDIS_MINIPORT_MAJOR_VERSION`.") and/or change this in the examples?

maxtruxa, 2015-08-03:
This would make things clear for any fellow developer who stumbles across this in the future.

jmrossy, 2015-08-04:
Good thinking! We've informed our docs team.
Thanks Max.

closed 2015-08-04 by jmrossy