microsoft/Windows-driver-samples #6: Maximum transfer length (HwMaxXferLen) calculation is off by 1 page size

tchaudev, 2015-07-01:
I believe the calculation for the HwMaxXferLen in the InitializeTransferPackets() function is off by 1 page size. Please see Windows-driver-samples/storage/class/classpnp/src/xferpkt.c line 70-73:

hwMaxPages = adapterDesc->MaximumPhysicalPages ? adapterDesc->MaximumPhysicalPages-1 : 0;
fdoData->HwMaxXferLen = MIN(adapterDesc->MaximumTransferLength, hwMaxPages << PAGE_SHIFT);
fdoData->HwMaxXferLen = MAX(fdoData->HwMaxXferLen, PAGE_SIZE);

jmrossy, 2015-07-09:
Thanks for pointing this out Trevor. You’re correct that the port driver needs to report a maximum number of pages that is off by one. 
In order to maintain backwards compatibility, we won’t be changing this Classpnp behavior in Windows 10.

closed 2015-07-09 by jmrossy

tchaudev, 2015-07-09:
Oh, so is it safe to say that MaximumPhysicalPages here is the reflected by the NumberOfPhysicalBreaks in the storage/miniports/storahci/src/entrypts.c line 575?

ConfigInfo->NumberOfPhysicalBreaks = 0x21;  // Since "NumberOfPhysicalBreaks" has been used in storage stack as the count of entries, use value of physical breaks plus one.

jmrossy, 2015-07-09:
Correct :)