microsoft/Windows-driver-samples #9: Would you please provide a usage example of the IOCTL_STORAGE_PROTOCOL_COMMAND by the DeviceIoControl?

sirosiro, 2015-08-07:
I have been looking for it but I couldn't find it anywhere, and I was tried to make it myself but it was not work correctly.

jmrossy, 2015-08-19:
Kouichi, thanks for bringing this to our attention. We're preparing a new doc page to describe the correct usage of this IOCTL. We'll share a link to that when it's live.

aym2k, 2016-01-18:
Any updates on the new doc?
I found this:
https://msdn.microsoft.com/en-us/library/windows/hardware/dn932068(v=vs.85).aspx

https://msdn.microsoft.com/en-us/library/windows/hardware/dn931814(v=vs.85).aspx

sirosiro, 2016-01-19:
I found already those documents when I opened this issue. Even if it was updated, I think it is not enough information to use this IOCTL.

I found out an important comment on the "ntddstor.h" file. (simler souce exist at "http://pastebin.com/VHxcWAbp".) according to that:
---
//
// Parameter for IOCTL_STORAGE_PROTOCOL_COMMAND
// Buffer layout: <STORAGE_PROTOCOL_COMMAND> <Command> [Error Info Buffer] [Data-to-Device Buffer] [Data-from-Device Buffer]
//
<snip>
//
// Additional notes when STORAGE_PROTOCOL_TYPE is ProtocolTypeNvme:
//  1.  When flag STORAGE_PROTOCOL_COMMAND_FLAG_ADAPTER_REQUEST is set, or the request is sent through adapter, namespace Id from "Command" field is used;
//      otherwise, the underneath driver should determine namespace Id from the device that receives the command.
//  2.  When a command fails, the "ErrorCode" field contains value from NVMe Completion Queue Entry - DW3 - Status Field.
//  3.  "CommandLength" field must have value of 64. e.g. STORAGE_PROTOCOL_COMMAND_LENGTH_NVME.
//  4.  "CommandSpecific" field must have value of either STORAGE_PROTOCOL_SPECIFIC_NVME_ADMIN_COMMAND, or STORAGE_PROTOCOL_SPECIFIC_NVME_NVM_COMMAND.
//  5.  When a command succeeds, field "FixedProtocolReturnData" may contain value from NVMe Completion Queue Entry - DW0.
//
---
And I tried to make a sample program with these information but it was not working correctly.
I can't find out more detailed description than the MSDN except the "ntddstor.h" file yet.
I need a usage example, or more detail document.

sirosiro, 2016-04-15:
Any update on this?

Does a new doc explained relations between the IOCTL_STORAGE_PROTOCOL_COMMAND and the Common Effect Log of NVMe v1.2 spec?
I think the document should explain a relation to NVMe v1.2 spec by the inbox driver (stornvme.sys). please refer below the link.

https://social.msdn.microsoft.com/Forums/windowshardware/en-US/5a2cda6f-83de-43d3-91c7-3fdc02c4971c/how-to-send-nvme-admin-command-in-windows-10?forum=wdk

And, Dose the document explain how to use extended buffers([Data-to-Device Buffer] [Data-from-Device Buffer]) with vender command?

psycholliu, 2016-06-27:
Hi, sirosiro
I am facing the same problem about IOCTL_STORAGE_PROTOCOL_COMMAND,have you solved this problem? and How? if you give me a usage of this ioctl, will help a lot. thanks.

sirosiro, 2016-06-27:
Hi, psycholliu,
Unfortunately, I don't resolve it yet. but I found out an information recently. that is below:

https://msdn.microsoft.com/en-us/library/windows/desktop/mt718131(v=vs.85).aspx

This page explaining to how to sending a vendor-specific command, but I don't tried it yet. 
I hope it'll be helpful for you. :)

psycholliu, 2016-06-28:
Hi sirosiro,
Thanks for your reply. I have gone though https://msdn.microsoft.com/en-us/library/windows/desktop/mt718131(v=vs.85).aspx, and  I have tried to send nvme IO Command(read command) with IOCTL_STORAGE_PROTOCOL_COMMAND, but failed with errorcode 87（parameter error）。thanks again！

side1out, 2016-11-17:
I too am stuck on the 0x87 error.  The bug whether it be documentation or otherwise has been reported to Microsoft.  It has just moved into the active state.    I too tried many variations of the sample code with no luck.  I get the 87 result with about anything I do which leads me to believe this command does not work at all meaning it is not coming down to command registers or buffers.

markg1, 2016-12-01:
IOCTL_STORAGE_PROTOCOL_COMMAND is intended for sending vendor-specific commands only. 
•Vendor Specific Admin Commands (C0h – FFh)
•Vendor Specific NVMe Commands (80h – FFh)
Sending an NVMe read command (Opcode 02h) will generate an error from this IOCTL.

Also your NVMe drive must support CEL.

wm1, 2017-03-16:
markg1 highlighted the problem in your experiments. And they are documented at https://msdn.microsoft.com/en-us/library/windows/desktop/mt718131(v=vs.85).aspx

closed 2017-03-16 by wm1

erikhugintech, 2017-04-01:
IOCTL_STORAGE_PROTOCOL_COMMAND is intended for sending vendor-specific commands only.
•Vendor Specific Admin Commands (C0h – FFh)
•Vendor Specific NVMe Commands (80h – FFh)

How can I get a namespace list? You block pass-through Identify, and you don't support Identify CNS 02 with IOCTL_STORAGE_QUERY_PROPERTY. 

Have you considered *not* blocking commands that don't do any damage, or do you actually have a rationale behind forcing the use of incomplete and needlessly complicated abstraction layers for completely harmless commands?

markg1, 2017-04-03:
I have not tried this, but based on nvme.h and provided info, I would suggest you try
following the _**Example: NVMe Identify query**_ at https://msdn.microsoft.com/en-us/library/windows/desktop/mt718131(v=vs.85).aspx. But replace the line:

protocolData->ProtocolDataRequestValue = NVME_IDENTIFY_CNS_CONTROLLER;
with 
protocolData->ProtocolDataRequestValue = NVME_IDENTIFY_CNS_ACTIVE_NAMESPACES;

Good luck.

RealCool1983, 2019-06-20:
> I have not tried this, but based on nvme.h and provided info, I would suggest you try
> following the _**Example: NVMe Identify query**_ at [...]

I try, but it can't work @@

mannanal, 2020-02-11:
Please refer to https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/nvme-features-supported-by-stornvme for any clarifications on command supported

Currently, IOCTL_STORAGE_QUERY_PROPERTY with identify option NVME_IDENTIFY_CNS_ACTIVE_NAMESPACES is not supported

Rozzon, 2024-08-13:
[resolved]
```
BOOL GetNVMeIdentifyData(HANDLE hDrive, BYTE* identifyData) {
    BOOL result = FALSE;
    PVOID buffer = NULL;
    ULONG bufferLength = 0;
    ULONG returnedLength = 0;

    PSTORAGE_PROPERTY_QUERY query = NULL;
    PSTORAGE_PROTOCOL_SPECIFIC_DATA protocolData = NULL;
    PSTORAGE_PROTOCOL_DATA_DESCRIPTOR protocolDataDescr = NULL;

    // Allocate buffer for query and NVMe data
    bufferLength = FIELD_OFFSET(STORAGE_PROPERTY_QUERY, AdditionalParameters) + sizeof(STORAGE_PROTOCOL_SPECIFIC_DATA) + NVME_MAX_LOG_SIZE;
    buffer = malloc(bufferLength);

    if (buffer == NULL) {
        printf("GetNVMeIdentifyData: Allocate buffer failed.\n");
        return FALSE;
    }

    // Initialize query data structure to get Identify Controller Data
    ZeroMemory(buffer, bufferLength);

    query = (PSTORAGE_PROPERTY_QUERY)buffer;
    protocolDataDescr = (PSTORAGE_PROTOCOL_DATA_DESCRIPTOR)buffer;
    protocolData = (PSTORAGE_PROTOCOL_SPECIFIC_DATA)query->AdditionalParameters;

    query->PropertyId = StorageAdapterProtocolSpecificProperty;
    query->QueryType = PropertyStandardQuery;

    protocolData->ProtocolType = ProtocolTypeNvme;
    protocolData->DataType = NVMeDataTypeIdentify;
    protocolData->ProtocolDataRequestValue = NVME_IDENTIFY_CNS_CONTROLLER;
    protocolData->ProtocolDataRequestSubValue = 0;
    protocolData->ProtocolDataOffset = sizeof(STORAGE_PROTOCOL_SPECIFIC_DATA);
    protocolData->ProtocolDataLength = NVME_MAX_LOG_SIZE;

    // Send request to get Identify Controller Data
    result = DeviceIoControl(
        hDrive,
        IOCTL_STORAGE_QUERY_PROPERTY,
        buffer,
        bufferLength,
        buffer,
        bufferLength,
        &returnedLength,
        NULL
    );

    if (!result) {
        DebugPrint("GetNVMeIdentifyData: DeviceIoControl failed with error %lu.\n", GetLastError());
        free(buffer);
        return FALSE;
    }

    // Validate and process the returned data
    ZeroMemory(buffer, bufferLength);
    query = (PSTORAGE_PROPERTY_QUERY)buffer;
    protocolDataDescr = (PSTORAGE_PROTOCOL_DATA_DESCRIPTOR)buffer;
    protocolData = (PSTORAGE_PROTOCOL_SPECIFIC_DATA)query->AdditionalParameters;

    query->PropertyId = StorageDeviceProtocolSpecificProperty;
    query->QueryType = PropertyStandardQuery;

    protocolData->ProtocolType = ProtocolTypeNvme;
    protocolData->DataType = NVMeDataTypeIdentify;
    protocolData->ProtocolDataRequestValue = NVME_IDENTIFY_CNS_CONTROLLER;
    protocolData->ProtocolDataRequestSubValue = 0;
    protocolData->ProtocolDataOffset = sizeof(STORAGE_PROTOCOL_SPECIFIC_DATA);
    protocolData->ProtocolDataLength = NVME_MAX_LOG_SIZE;

    // Send request again to get the actual data
    result = DeviceIoControl(
        hDrive,
        IOCTL_STORAGE_QUERY_PROPERTY,
        buffer,
        bufferLength,
        buffer,
        bufferLength,
        &returnedLength,
        NULL
    );

    if (!result) {
        DebugPrint("GetNVMeIdentifyData: DeviceIoControl failed with error %lu.\n", GetLastError());
        free(buffer);
        return FALSE;
    }

    if ((protocolDataDescr->Version != sizeof(STORAGE_PROTOCOL_DATA_DESCRIPTOR)) ||
        (protocolDataDescr->Size != sizeof(STORAGE_PROTOCOL_DATA_DESCRIPTOR))) {
        DebugPrint("GetNVMeIdentifyData: Data descriptor header not valid.\n");
        free(buffer);
        return FALSE;
    }

    protocolData = &protocolDataDescr->ProtocolSpecificData;

    if ((protocolData->ProtocolDataOffset < sizeof(STORAGE_PROTOCOL_SPECIFIC_DATA)) ||
        (protocolData->ProtocolDataLength < NVME_MAX_LOG_SIZE)) {
        DebugPrint("GetNVMeIdentifyData: ProtocolData Offset/Length not valid.\n");
        free(buffer);
        return FALSE;
    }

    // Copy raw NVMe Identify Controller Data to output buffer
    PCHAR rawData = (PCHAR)protocolData + protocolData->ProtocolDataOffset;

    memcpy(identifyData, rawData, protocolData->ProtocolDataLength);
    DebugPrint("GetNVMeIdentifyData: Identify Controller Data succeeded.\n");
    DebugPrintIdentifyData(identifyData, protocolData->ProtocolDataLength);
    free(buffer);
    return TRUE;
}
```
I get identify controller data succeeded using IOCTL_STORAGE_QUERY_PROPERTY