microsoft/Windows-driver-samples #16: USBView doesn't build because of usbschema.hpp

Tails86, 2015-09-17:
I can't get USBView to build because there are missing definitions within usbschema.hpp. For instance,  deviceIsOperatingAtSuperSpeedPlusOrHigher variable is not defined and UsbBillboardCapabilityDescriptorType class is not defined. When I try to run "xsd.exe /language:CPP /c /order /namespace:Microsoft.Kits.Samples.Usb" the xsd application just spits out options list. What do I need to do to get this to compile?

Tails86, 2015-09-17:
It seems like I'm missing the xsd schema file in order to regenerate usbschema.hpp file myself.

DavidEGrayson, 2015-09-24:
I just got usbview to build in MSYS2, and added a package for it: https://github.com/Alexpux/MINGW-packages/tree/master/mingw-w64-usbview

I had some of the same problems you had, and I solved them by building an older version of usbview that can be found here:

https://code.msdn.microsoft.com/windowshardware/USBView-sample-application-e3241039/file/51333/38/USBView%20sample%20application.zip

barrygolden, 2015-10-05:
Updated usbschema.hpp to fix compile error.

closed 2015-10-05 by barrygolden