microsoft/Windows-driver-samples #19: The README for avstream / samplemft0 is incomplete

timkeosa, 2015-09-23:
The "Driver MFT" sample originally on MSDN contained a comprehensive README file detailing the complex steps necessary to install the MFT0 module. This README contains none of that information, and now that the original sample on MSDN has been pulled, this important information is unavailable.

Please update the README for this version of the MFT0 sample to include the original MSDN documentation and also fix the dead links.

MukeshA1, 2019-02-11:
Hi,
Any pointers where I can check original MSDN documentation to get more insight. I am facing error in installing the SampleMft0.dll.
Thanks

v-junyli, 2024-11-25:
We are closing out this issue that unfortunately have not been addressed for a long time.  If you have data that issue remains, then feel free to reactivate

closed 2024-11-25 by v-junyli