microsoft/Windows-driver-samples #20: Request: source code for inf2cat and signtool

DavidEGrayson, 2015-09-24:
Since [inf2cat](https://msdn.microsoft.com/en-us/library/windows/hardware/ff547089(v=vs.85).aspx) and [signtool](https://msdn.microsoft.com/en-us/library/windows/hardware/ff551778(v=vs.85).aspx) are important tools used to make drivers, I think the Windows driver community would benefit from seeing how they work.  I would request that you add the source code of those two utilities in this repository.  (Or if the source code is available somewhere else, please let me know.  I could not find it.)

wm1, 2017-03-15:
Other tool like devcon is useful because IHV might need to write their own setup utility; but I cannot think of any user case when IHV need to write another signtool.

closed 2017-03-15 by wm1

DavidEGrayson, 2017-04-07:
I can think of a big reason to make signtool open source.  The "signtool verify" feature is supposed to check whether your signature is good, but there are many cases where "signtool verify" shows no errors and yet developers still run into problems when trying to use the signed files (e.g. Code 52 when loading a driver).  I'm helping someone over email right who is in that situation.  I wrote this [big article](http://www.davidegrayson.com/signing) with the [requirements](http://www.davidegrayson.com/signing/#reqs) for a successful signature and I think signtool doesn't check most of those requirements.  I know Vista is almost dead, but "signtool" never warned about how kernel modules have to be signed with SHA-1 to work in Windows Vista.  I have not tested this recently but I don't think it provides any warnings against using SHA-1 now that it is being phased out.