microsoft/Windows-driver-samples #21: I can not find sysvad.cat and sysvad.inf after sucessfully built sysavd project

wanzhang, 2015-10-05:
I follow the build steps in Readme.md, but can't find sysvad.inf file after build successfully,Only the following files can be found in Debug folder:

2015/10/05  08:49           615,936 KeywordDetectorContosoAdapter.dll
2015/10/05  09:09           240,008 PhoneAudioSample.sys
2015/10/05  08:51           239,616 PropPageExt.dll
2015/10/05  08:51           167,424 SwapAPO.dll
2015/10/05  09:09           214,264 TabletAudioSample.sys

I don't know how to install this audio driver sample in my target computer. Do I miss some settings in my building process, or the Readme document has not been updated? Please help.

mvaneerde, 2015-10-08:
There's a problem with the package.vcxproj file. Here's how to work around it:

Right-click the "package" project in Visual Studio.
Choose "Add Existing Item".
Go up a level (..) and choose phoneaudiosample.inf and tabletaudiosample.inf.

Now rebuild the solution. The output will have the suitably-stamped .inf files, and a signed .cat file.

wanzhang, 2015-10-09:
Thanks for reply.Now I can get xx.inf and xx.cat file in Debug folder, but when I installed tabletaudiosample.inf on my target computer, it seems something wrong.
My installation cmd is:

E:\data\output\x64>devcon.exe install tabletaudiosample.inf Root\sysvad_TabletAu
dioSample

and I find a "!" in "Virtual Audio Device(WDM) - Tablet Sample" item in device manager, and get this information:

    The Windows cannot load the device driver for this hardware. The driver may be corrupted or missing. (Code 39)

Any suggestion?

mvaneerde, 2015-10-09:
tabletaudiosample.inf has a DeviceInstall32 section so you can use "pnputil.exe -a tabletaudiosample.inf" to install it; the physical device object is created by Windows when the driver is added to the store.
C:\Windows\inf\setupapi.dev.log will have more detailed error messages on why the driver is "corrupted or missing".

mvaneerde, 2015-10-28:
Fixed in https://github.com/Microsoft/Windows-driver-samples/commit/2c9b5b696dc2c396e6ffc9ec721d8085f279a114 - thanks Adam!

ralusa, 2016-05-11:
I was download the WDK10 and tried to build the sysvad. Initially it shows error such as C2220, C4595. So, I made change in the properties that is "Treat errors as warnings - NO", after this change it successfully builds and generated required dll and cat file.

Then I have tried to install using "pnputil.exe -a tabletaudisample.inf". Installation success but device is added to "other devices category in device manager ".

pnputil.exe -a tabletaudisample.inf comand prompt logs : 
Processing inf : tabletaudiosample.inf
Driver package added successfully.
Published name : oem37.inf
Total attempted: 1
Number successfully imported: 1

setupapi.dev.log

[SetupCopyOEMInf - C:\install_drivers\tabletaudiosample.inf]
Section start 2016/05/11 10:42:26.565
cmd: pnputil.exe -a tabletaudiosample.inf 
inf: Copy style: 0x00000000
sto: {Setup Import Driver Package: C:\install_drivers\tabletaudiosample.inf} 10:42:26.745
inf: Provider: TODO-Set-Provider
inf: Class GUID: {4d36e96c-e325-11ce-bfc1-08002be10318}
inf: Driver Version: 05/10/2016,17.11.14.792
inf: Catalog File: sysvad.cat
sto: {Copy Driver Package: C:\install_drivers\tabletaudiosample.inf} 10:42:26.763
sto: Driver Package = C:\install_drivers\tabletaudiosample.inf
sto: Flags = 0x00000007
sto: Destination = C:\Users\MMTEST~1\AppData\Local\Temp{97fc79c7-9297-c047-b801-c161ae7bb7f4}
sto: Copying driver package files to 'C:\Users\MMTEST~1\AppData\Local\Temp{97fc79c7-9297-c047-b801-c161ae7bb7f4}'.
flq: Copying 'C:\install_drivers\sysvad.cat' to 'C:\Users\MMTEST~1\AppData\Local\Temp{97fc79c7-9297-c047-b801-c161ae7bb7f4}\sysvad.cat'.
flq: Copying 'C:\install_drivers\tabletaudiosample.inf' to 'C:\Users\MMTEST~1\AppData\Local\Temp{97fc79c7-9297-c047-b801-c161ae7bb7f4}\tabletaudiosample.inf'.
sto: {Copy Driver Package: exit(0x00000000)} 10:42:26.798
pol: {Driver package policy check} 10:42:26.825
pol: {Driver package policy check - exit(0x00000000)} 10:42:26.826
sto: {Stage Driver Package: C:\Users\MMTEST~1\AppData\Local\Temp{97fc79c7-9297-c047-b801-c161ae7bb7f4}\tabletaudiosample.inf} 10:42:26.827
inf: {Query Configurability: C:\Users\MMTEST~1\AppData\Local\Temp{97fc79c7-9297-c047-b801-c161ae7bb7f4}\tabletaudiosample.inf} 10:42:26.838
inf: Driver package is fully isolated.
inf: Driver package 'tabletaudiosample.inf' is configurable.
inf: {Query Configurability: exit(0x00000000)} 10:42:26.842
flq: Copying 'C:\Users\MMTEST~1\AppData\Local\Temp{97fc79c7-9297-c047-b801-c161ae7bb7f4}\sysvad.cat' to 'C:\WINDOWS\System32\DriverStore\Temp{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\sysvad.cat'.
flq: Copying 'C:\Users\MMTEST~1\AppData\Local\Temp{97fc79c7-9297-c047-b801-c161ae7bb7f4}\tabletaudiosample.inf' to 'C:\WINDOWS\System32\DriverStore\Temp{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\tabletaudiosample.inf'.
sto: {DRIVERSTORE IMPORT VALIDATE} 10:42:26.857
sig: {_VERIFY_FILE_SIGNATURE} 10:42:26.872
sig: Key = tabletaudiosample.inf
sig: FilePath = C:\WINDOWS\System32\DriverStore\Temp{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\tabletaudiosample.inf
sig: Catalog = C:\WINDOWS\System32\DriverStore\Temp{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\sysvad.cat
! sig: Verifying file against specific (valid) catalog failed! (0x800b0109)
! sig: Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
sig: {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 10:42:26.878
sig: {_VERIFY_FILE_SIGNATURE} 10:42:26.878
sig: Key = tabletaudiosample.inf
sig: FilePath = C:\WINDOWS\System32\DriverStore\Temp{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\tabletaudiosample.inf
sig: Catalog = C:\WINDOWS\System32\DriverStore\Temp{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\sysvad.cat
! sig: Verifying file against specific Authenticode(tm) catalog failed! (0x800b0109)
! sig: Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
sig: {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 10:42:26.882
! sig: Driver package catalog file certificate does not belong to Trusted Root Certificates, but user wants to install anyway.
sto: {DRIVERSTORE IMPORT VALIDATE: exit(0x00000000)} 10:42:29.041
sig: Signer Score = 0x80000000
sig: Signer Name = 
sto: {DRIVERSTORE IMPORT BEGIN} 10:42:29.043
bak: Create system restore point:
bak: Description = Device Driver Package Install: TODO-Set-Provider Sound, video and game controllers
bak: Time = 0ms
bak: Status = 0x00000000 (SUCCESS)
sto: {DRIVERSTORE IMPORT BEGIN: exit(0x00000000)} 10:42:29.054
cpy: {Copy Directory: C:\WINDOWS\System32\DriverStore\Temp{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}} 10:42:29.055
cpy: Target Path = C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_11c7c5be0f5048f7
cpy: {Copy Directory: exit(0x00000000)} 10:42:29.056
idb: {Register Driver Package: C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_11c7c5be0f5048f7\tabletaudiosample.inf} 10:42:29.057
idb: Created driver package object 'tabletaudiosample.inf_amd64_11c7c5be0f5048f7' in DRIVERS database node.
idb: Created driver INF file object 'oem37.inf' in DRIVERS database node.
[... 6 lines cut ...]
sto: Flushed driver database node 'DRIVERS'. Time = 78 ms
sto: Flushed driver database node 'SYSTEM'. Time = 31 ms
idb: {Publish Driver Package: exit(0x00000000)} 10:42:29.187
sto: {DRIVERSTORE IMPORT END} 10:42:29.197
sig: Installed catalog 'sysvad.cat' as 'oem37.cat'.
[... cut ...]
dvi: {Install Root Device: exit(0x00000000)} 10:42:29.248
sto: {DRIVERSTORE IMPORT END: exit(0x00000000)} 10:42:29.249
sto: {Stage Driver Package: exit(0x00000000)} 10:42:29.250
sto: {Setup Import Driver Package - exit (0x00000000)} 10:42:29.259
inf: Driver Store Path: C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_11c7c5be0f5048f7\tabletaudiosample.inf
inf: Published Inf Path: C:\WINDOWS\INF\oem37.inf
<<< Section end 2016/05/11 10:42:29.271
<<< [Exit status: SUCCESS]

mvaneerde, 2016-05-11:
> Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider

This is the problem - your test machine doesn't trust your signing certificate.

ralusa, 2016-05-11:
Thanks mvaneerde, 
How do I make the test machine to trust the signing certificate ?

mvaneerde, 2016-05-11:
See the instructions for installing test drivers linked from the README

https://github.com/Microsoft/Windows-driver-samples/blob/master/audio/sysvad/README.md

ralusa, 2016-05-11:
Thanks for your suggestion,
I followed the steps mentioned in the readme.md and now in oem32.inf info is showing the signer name 

Published name :            oem37.inf
Driver package provider :   TODO-Set-Provider
Class :                     Sound, video and game controllers
Driver date and version :   05/11/2016 16.53.5.175
Signer name :               WDKTestCert mmtestuser,131072637947324303

but still i found that "Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.".  What could be the reason ?

>>>  [SetupCopyOEMInf - C:\Tools\tabletaudiosample.inf]
>>>  Section start 2016/05/11 17:02:01.251
      cmd: pnputil.exe  -a tabletaudiosample.inf
     inf: Copy style: 0x00000000
     sto: {Setup Import Driver Package: C:\Tools\tabletaudiosample.inf} 17:02:01.265
     inf:      Provider: TODO-Set-Provider
     inf:      Class GUID: {4d36e96c-e325-11ce-bfc1-08002be10318}
     inf:      Driver Version: 05/11/2016,16.53.5.175
     inf:      Catalog File: sysvad.cat
     sto:      {Copy Driver Package: C:\Tools\tabletaudiosample.inf} 17:02:01.286
     sto:           Driver Package = C:\Tools\tabletaudiosample.inf
     sto:           Flags          = 0x00000007
     sto:           Destination    = C:\Users\MMTEST~1\AppData\Local\Temp\{fba2c9d0-9b99-424f-a93c-2bc88701965e}
     sto:           Copying driver package files to 'C:\Users\MMTEST~1\AppData\Local\Temp\{fba2c9d0-9b99-424f-a93c-2bc88701965e}'.
     flq:           Copying 'C:\Tools\sysvad.cat' to 'C:\Users\MMTEST~1\AppData\Local\Temp\{fba2c9d0-9b99-424f-a93c-2bc88701965e}\sysvad.cat'.
     flq:           Copying 'C:\Tools\tabletaudiosample.inf' to 'C:\Users\MMTEST~1\AppData\Local\Temp\{fba2c9d0-9b99-424f-a93c-2bc88701965e}\tabletaudiosample.inf'.
     sto:      {Copy Driver Package: exit(0x00000000)} 17:02:01.316
     pol:      {Driver package policy check} 17:02:01.322
     pol:      {Driver package policy check - exit(0x00000000)} 17:02:01.323
     sto:      {Stage Driver Package: C:\Users\MMTEST~1\AppData\Local\Temp\{fba2c9d0-9b99-424f-a93c-2bc88701965e}\tabletaudiosample.inf} 17:02:01.323
     inf:           {Query Configurability: C:\Users\MMTEST~1\AppData\Local\Temp\{fba2c9d0-9b99-424f-a93c-2bc88701965e}\tabletaudiosample.inf} 17:02:01.335
     inf:                Driver package is fully isolated.
     inf:                Driver package 'tabletaudiosample.inf' is configurable.
     inf:           {Query Configurability: exit(0x00000000)} 17:02:01.338
     flq:           Copying 'C:\Users\MMTEST~1\AppData\Local\Temp\{fba2c9d0-9b99-424f-a93c-2bc88701965e}\sysvad.cat' to 'C:\WINDOWS\System32\DriverStore\Temp\{a9216593-5821-424b-99fb-ce367e2edb5d}\sysvad.cat'.
     flq:           Copying 'C:\Users\MMTEST~1\AppData\Local\Temp\{fba2c9d0-9b99-424f-a93c-2bc88701965e}\tabletaudiosample.inf' to 'C:\WINDOWS\System32\DriverStore\Temp\{a9216593-5821-424b-99fb-ce367e2edb5d}\tabletaudiosample.inf'.
     sto:           {DRIVERSTORE IMPORT VALIDATE} 17:02:01.356
     sig:                {_VERIFY_FILE_SIGNATURE} 17:02:01.368
     sig:                     Key      = tabletaudiosample.inf
     sig:                     FilePath = C:\WINDOWS\System32\DriverStore\Temp\{a9216593-5821-424b-99fb-ce367e2edb5d}\tabletaudiosample.inf
     sig:                     Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{a9216593-5821-424b-99fb-ce367e2edb5d}\sysvad.cat
!    sig:                     Verifying file against specific (valid) catalog failed! (0x800b0109)
!    sig:                     Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:                {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 17:02:01.368
     sig:                {_VERIFY_FILE_SIGNATURE} 17:02:01.368
     sig:                     Key      = tabletaudiosample.inf
     sig:                     FilePath = C:\WINDOWS\System32\DriverStore\Temp\{a9216593-5821-424b-99fb-ce367e2edb5d}\tabletaudiosample.inf
     sig:                     Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{a9216593-5821-424b-99fb-ce367e2edb5d}\sysvad.cat
     sig:                     Success: File is signed in Authenticode(tm) catalog.
     sig:                     Error 0xe0000241: The INF was signed with an Authenticode(tm) catalog from a trusted publisher.
     sig:                {_VERIFY_FILE_SIGNATURE exit(0xe0000241)} 17:02:01.384
     sto:           {DRIVERSTORE IMPORT VALIDATE: exit(0x00000000)} 17:02:01.388
     sig:           Signer Score = 0x0F000000
     sig:           Signer Name  = WDKTestCert mmtestuser,131072637947324303
     sto:           {DRIVERSTORE IMPORT BEGIN} 17:02:01.390
     sto:           {DRIVERSTORE IMPORT BEGIN: exit(0x00000000)} 17:02:01.390
     cpy:           {Copy Directory: C:\WINDOWS\System32\DriverStore\Temp\{a9216593-5821-424b-99fb-ce367e2edb5d}} 17:02:01.391
     cpy:                Target Path = C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_a829bf2a622c5687
     cpy:           {Copy Directory: exit(0x00000000)} 17:02:01.393
     idb:           {Register Driver Package: C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_a829bf2a622c5687\tabletaudiosample.inf} 17:02:01.394
     idb:                Created driver package object 'tabletaudiosample.inf_amd64_a829bf2a622c5687' in DRIVERS database node.
[... cut ...]
     dvi:                {Install Root Device: exit(0x00000000)} 17:02:01.538
     sto:           {DRIVERSTORE IMPORT END: exit(0x00000000)} 17:02:01.539
     sto:      {Stage Driver Package: exit(0x00000000)} 17:02:01.540
     sto: {Setup Import Driver Package - exit (0x00000000)} 17:02:01.549
     inf: Driver Store Path: C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_a829bf2a622c5687\tabletaudiosample.inf
     inf: Published Inf Path: C:\WINDOWS\INF\oem37.inf
<<<  Section end 2016/05/11 17:02:01.561
<<<  [Exit status: SUCCESS]

mvaneerde, 2016-05-11:
You need to tell the machine to trust the root certificate used to sign the driver. See https://msdn.microsoft.com/windows/hardware/drivers/develop/preparing-a-computer-for-manual-driver-deployment

It's important to import the certificate to the Local Machine store - it does not good to import to Current User. https://blogs.msdn.microsoft.com/matthew_van_eerde/2015/09/02/poking-into-the-trusted-root-certification-authorities-certificate-stores-for-the-current-user-and-the-local-machine/

ralusa, 2016-05-18:
Thank you. Its working.

jlehtniemi-broadsoft, 2016-09-07:
The original problem with missing `sysvad.cat` file persists despite of fix in 2c9b5b6.
(I'm building from 687b274 because of #76)

closed 2016-11-04 by wanzhang