microsoft/Windows-driver-samples #22: tabletaudiosample.inf registers APOs incorrectly

mvaneerde, 2015-10-08:
There are a handful of lines like this in tabletaudiosample.inf:
HKCR,CLSID\%FX_STREAM_CLSID%\InProcServer32,,0x0002000,%%SystemRoot%%\System32\swapapo.dll

The "0x0002000" is an error - it should be 0x00020000 (four zeros after the 2.) This error causes the swapapo.dll binaries not to be registered with COM.

To fix this, do a Find/Replace to change all the 0x0002000 to 0x00020000 in tabletaudiosample.inf.

wanzhang, 2015-10-09:
Install fail. A "!" icon still can be found in "Virtual Audio Device(WDM) - Tablet Sample" item.

mvaneerde, 2015-10-28:
Fixed in https://github.com/Microsoft/Windows-driver-samples/commit/2c9b5b696dc2c396e6ffc9ec721d8085f279a114 - thanks Adam!

closed 2015-10-28 by mvaneerde