microsoft/Windows-driver-samples #28: How to program to update driver software from hidscanner.dll to hidclass.sys for HID Barcode Scanner(usage page is 0x008C) on Windows 10

namizhipianren, 2015-10-23:
Hi there,

the hidscanner.dll is loaded automatically for HID Barcode Scanner(usage page is 0x008C) on Windows 10, but our application needs to use hid.lib so hidclass.sys should be used as default. 

Since it is uncomfortable for customer to update the driver software manually, so is it possible to force Windows 10 to load hidclass.sys by any configuration of the system, or is it possible and how to program to update driver software?

namizhipianren, 2015-10-23:
Learning from DevCon.exe does help.

mchen-msft, 2015-11-13:
FYI, you should still be able to use Hid.lib with hidscanner.dll loaded.  The hidscanner driver is loaded on top of the hid device created by hidclass.sys, and the hidscanner driver normally passes through requests like a filter driver.

Just wondering, is your application having issues enumerating the scanner?  Did your application work on Windows 8.1 and it no longer works on 10?

closed 2018-01-16 by wm1