microsoft/Windows-driver-samples #31: XPSDrvSample

wilk1, 2015-11-11:
Hi,
I get the following error message when I try and build the "XPSDrv Driver and Filter Sample" project. Followed the instructions in the Readme guide, but I also get exact same error message when Building the built-in "Print Driver v4" template. Not sure if it's something I have to configure within each project or on my machine.
Using Visual Studio 2015 Express.
Inf2Cat Tool Output:
.............................
Signability test failed.

Errors:
22.9.1: xdwscrgb.icc in [colorprofiles] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdcmykprinter.icc in [colorprofiles] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl.ini in [configplugin] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdnames.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdwmark.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdbook.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdcolman.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdnup.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdpgscl.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl-pipelineconfig.xml in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.

Warnings:
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntx86] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntia64] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntamd64] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntx86.6.0] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntia64.6.0] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntamd64.6.0] of \xdsmpl.inf

Autismleo, 2017-09-06:
aaaaa

stephenegriffin, 2017-12-22:
Same as #123

PR #208 (2018-07-19): Update README.md
Update build instructions. Resolves #123. Resolves #197. Resolves #206. Resolves #31.
commit c97def4e (2018-07-19): Update README.md

Update build instructions. Resolves #123. Resolves #197. Resolves #206. Resolves #31.
closed 2018-07-19 by NeoAdonis