microsoft/Windows-driver-samples #34: Sample/Instructions to install NetService (NDIS Filter Driver) on Windows IoT Core

charlehsin, 2015-11-20:
I know that there is a simple driver example to install a device driver on IoT Core. I cannot find any information about how to install a NetService type NDIS filter driver on IoT Core. I have trouble installing a NDIS Filter Driver on IoT Core, which is quite different from a general device driver.

My NDIS filter driver is basically from VS2015 NDIS filter driver sample. I already successfully put it on a target Win 10 Enterprise PC and can see its trace messages via VS2015 Kernel debugger window from Development computer. 

I was also trying to put it on Windows IoT Core, and my hardware is Minnowboard Max. My steps are:
1. Follow https://ms-iot.github.io/content/en-US/win10/samples/DriverLab4.htm to just set up the provisioning on the target IoT Core from VS2015 on dev computer.
2. Follow https://channel9.msdn.com/Blogs/WinHEC/Creating-Universal-Drivers-with-WDK-10 to prepare for the mobile package (.cab file). Need to enter the required info in Package.pkg.xml file.
Build the x86 debug version.
3. Follow https://channel9.msdn.com/Blogs/WinHEC/Creating-Universal-Drivers-with-WDK-10 to deploy the .cab (online .cab package installation/update) on IoT Core from VS2015.

However, after connected via Kernel Debugger and set Kd_DEFAULT_MASK = 0xF,
 cannot see any filter driver’s trace messages. 
“devcon listclass NetService” returns empty.

Can anyone give me any instructions or hint? (This issue
 is only for IoT Core. I successfully made it work on Windows 10 Enterprise already.)

jordanrh1, 2015-12-01:
NDIS filter drivers are not supported on IoT Core since there is no facility to register the filter with the network driver binding engine. The work to bring this scenario online is being evaluated but there are no definite plans yet.

charlehsin, 2015-12-01:
ok. Thanks.

closed 2015-12-01 by charlehsin

druske, 2020-01-30:
Sorry for responding to an old closed issue, but the context is relevant. It has been more than 4 years. Has IoT Core budged at all in its support for NDIS filter or protocol drivers?