microsoft/Windows-driver-samples #36: Is there any guideline to transfer audio samples from virtual sound card to real one?

wanzhang, 2015-11-26:
I‘m considering to develop a virtual sound card driver, which could capture all audio data. But at the same time, the audio data need to play in a real sound card driver. The question is how could I transfer audio samples from my virtual sound card to real sound card?  Is it possible to finish it in Windows 10 and what  documents should I read?

wanzhang, 2015-11-26:
I find Audio Repeater could do my job, but I do really want to know what technology behind it.

Adonis-Song, 2022-12-13:
@wanzhang Hi. I met same question. Do you have any ideas?

v-junyli, 2024-11-25:
We are closing out this issue that unfortunately have not been addressed for a long time.  If you have data that issue remains, then feel free to reactivate.

closed 2024-11-25 by v-junyli