microsoft/Windows-driver-samples #39: C28118 in EvtCleanupCallback of WDF_DRIVER_CONFIG

chm-tm, 2016-01-05:
The `EvtCleanupCallback` implementations of at least the PLX9x5x and the PciDrv samples use the `PAGED_CODE()` macro, which will issue the [undocumented](https://social.msdn.microsoft.com/Forums/de-DE/ba0e4557-05a7-42a0-a960-cf2ded57ecfb/driver-analysis-undocumented-warning-c28118?forum=wdk) C28118 code analysis warning.

The `EvtCleanupCallback` function documentation says the following:
> Typically, the framework calls the EvtCleanupCallback callback function at IRQL <= DISPATCH_LEVEL. However, the framework calls the callback function at IRQL = PASSIVE_LEVEL in the following situations:
> The object's handle type is WDFDEVICE, WDFDRIVER, WDFDPC, WDFINTERRUPT, WDFIOTARGET, WDFQUEUE, WDFSTRING, WDFTIMER, or WDFWORKITEM.
> ...

This indicates C28118 is a false positive here, or the documentation deviates from the declaration of the `EVT_WDF_DEVICE_CONTEXT_CLEANUP` typedef.

Based on the answer, the samples could be updated accordingly.

The same goes for EvtDeviceContextCleanup.

chm-tm, 2016-01-05:
I suggest to insert pragmas where appropriate:
```c
#pragma warning ( disable: 28118 )
```

wm1, 2017-03-15:
Each warning need be analyzed individually. When it is documented to be passive level, add the following; otherwise you should remove `PAGED_CODE` macro.
```
    // EvtCleanupCallback for WDFDRIVER is always called at PASSIVE_LEVEL
    //
    _Analysis_assume_(KeGetCurrentIrql() == PASSIVE_LEVEL);
```

closed 2017-03-15 by wm1

chm-tm, 2017-03-16:
@wm1: Thanks for the code. I just saw the `__analysis_assume` macro which however indicates I shouldn't be calling a function in the expression:
```C
/*
* Please do not put function calls in the expression because this is not
* supported by all tools:
*  __analysis_assume(GetObject () != NULL); // DO NOT DO THIS
*/
```
The SAL version you recommend does not seem to have such a restriction. Actually, I didn't find any documentation for the `_Analysis_assume_` macro. Do you have any insights on the macro, and when to use which version of these macros?

wm1, 2017-03-16:
Find a better approach: `_IRQL_limited_to_(PASSIVE_LEVEL)`

This is from Windows Kits\10\Include\10.0.14393.0\shared\\`DriverSpecs.h`:
```
    // Callback with high IRQL level will never actually be called above 
    // 'level'
    #define _IRQL_limited_to_(level)                                      \
        ASSERT(KeGetCurrentIrql() <= level);                              \
        _Analysis_assume_(KeGetCurrentIrql() <= level);
```

chm-tm, 2017-03-17:
Ah, interesting. While this macro seems to be undocumented in MSDN either, it documents the usage of the `_Analysis_assume_` macro.