microsoft/Windows-driver-samples #44: RAMDisk Storage Driver Sample

marinkobabic, 2016-01-20:
Have installed the ramdrive on Windows 10 x64. I used the manual installation approach. The device was mounted and I have been able to see the drive R in explorer. When running the cmd console as administrator  and trying to switch to the drive R: "incorrect function" message is displayed.

tchaudev, 2016-02-18:
Did you ever figured out this issue? I'm having a similar problem, except I don't even see the drive 'R' in explorer. When using the Command Prompt I'm also getting "incorrect function", and a "DriveNotFoundException" when using PowerShell.

marinkobabic, 2016-02-19:
In the example FILE_SYSTEM_DISK is used. So when the drive gets the first request it tries to mount the volume. The mounting of the volume is not handled properly in the example and that is the reason we get this error message. 

On the web a have found the previous version of ramdisk for x86 operating system and this example is working as expected. It has been built for windows 2000/xp.

tchaudev, 2016-02-19:
Thanks for the info.

tchaudev, 2016-03-04:
I updated the RAMDisk driver sample so that drive "R:" can be accessed via command line. Please see pull request #53, if you're interested.

wzydhek, 2016-05-22:
I have the latest version and have it working correctly in my hyper-v machines, but I still get the incorrect function when using it on my host machine (also windows 10 x64).

cprogrammer1994, 2017-10-21:
I have a virtual machine with Windows x64. the driver installs properly with the devcon.exe. The `R:` drive appears. When I try to open it I get an "Access Denied" error. Running a cmd as administrator and running `R:` I get "Incorrect function."

What is the solution for this?

tchaudev, 2017-10-21:
See pull request #53

cprogrammer1994, 2017-10-21:
@tchaudev I forked your repo. I had some build errors in the inx file. I fixed them. It is not working either :(
I always test using a snapshot of a clean Windows install.

Did you test on Windows 10 x64?

tchaudev, 2017-10-21:
Yes, I tested on Windows 10 x64, but that was a long time ago. It is possible that the fix is out dated. That said, the problem was pretty simple. The reason why you get access denied is because the Ramdisk driver was not handling IOCTL queries that was introduced in Windows 10. If you update the code to properly handle those IOCTL, then everything should work.

cprogrammer1994, 2017-10-21:
Thank you! At least I know where to start. :)

chebotarevmichael, 2018-04-23:
@tchaudev, could you open source code of your sample for RAMdisk? It was not found in samples :(

tchaudev, 2018-04-23:
@chebotarevmichael you'll have to get the code from my fork: https://github.com/tchaudev/Windows-driver-samples

wonderkun, 2019-04-24:
> Thank you! At least I know where to start. :)
I have the same problem with you . I clone the ramdisk and build it  without any errors. But when it run on windows 10 x64,The R:  DriveLetter appear, but no one can access it except the administrator. I can only read and write it use command prompt as admin . How to fix it?

wonderkun, 2019-04-24:
@cprogrammer1994 
I run ramdisk on windows10 x64 without any errors. But no one can access it except the administrator.  I can only read and write it with Command Prompt window as Administrator. How to fix it ? please help me ,thank you very much.

tchaudev, 2019-04-28:
@wonderkun, I haven't tried it, but check out Jichao comment: https://github.com/Microsoft/Windows-driver-samples/pull/53#issuecomment-282203661

wonderkun, 2019-04-28:
@tchaudev  Thank you very much.

wonderkun, 2019-04-28:
@tchaudev  Thank you very much. I fix it.