microsoft/Windows-driver-samples #45: error 1308: Found legacy AddReg operation defining class installers (Installer32)

tchaudev, 2016-01-27:
I found this error trying to build the **Toaster Driver Sample** (Windows-driver-samples\general\toaster\toastDrv). According to the [INF Validation Errors and Warnings](https://msdn.microsoft.com/en-us/library/windows/hardware/dn929320(v=vs.85).aspx) page, this is a **Universal INF errors**. To get around this issue, I changed the target platform set from **Universal** to **Desktop**. Is this the proper thing to do, or is it better to fix the **.inx** files?

Shyamal-V, 2016-01-28:
Thanks for reporting this. We'll fix this in a subsequent release.

If the driver you are building targets Desktop based editions of Windows, changing the target platform to Desktop is ok. If you're building the driver and want it to be [universal](https://msdn.microsoft.com/en-us/windows-drivers/develop/getting_started_with_universal_drivers) then you should remove the class installer reference from the INX files. You can find more information on Universal drivers [here](https://msdn.microsoft.com/en-us/windows-drivers/develop/getting_started_with_universal_drivers).

tchaudev, 2016-01-28:
Thanks for the info.

sirotnikov, 2016-08-08:
this is still an issue

Shyamal-V, 2016-08-10:
@sirotnikov, there hasn't been an update/refresh of the samples for the Windows 10 1607 (anniversary update) release as yet. I've followed up with the folks that will be performing the update and will let you know when I have more information. Thanks.

Shyamal-V, 2016-08-12:
The samples were updated yesterday. Please try again.

tchaudev, 2016-08-12:
The updates broke the build completely. Before, I could have at least build for the Desktop platform.

Shyamal-V, 2016-08-12:
Could you please report that as a new issue with details of the build issues you are seeing. I'll have the team that manages the kits/samples investigate it.

tchaudev, 2016-08-12:
Sure. I'll do that when I have more time to look into the errors.

closed 2016-08-12 by tchaudev