microsoft/Windows-driver-samples #46: sysvad installation fails

tolnaisz, 2016-02-03:
I have followed the readme.md to build sysvad. First thing I have noticed, that it didn't create a cat file. Anyway, I have copied the tabletaudiosample.inf to the package directory and copied that to a freshly installed Win10 (build 10240) virtual machine.

I have executed the "bcdedit /set TESTSIGNING ON" command and rebooted.

Then I run the "devcon.exe install package\tabletaudiosample.inf Root\sysvad_TabletAudioSample" command and received the following:
Device node created. Install is complete when drivers are installed...
Updating drivers for Root\sysvad_TabletAudioSample from C:\Users\m\Desktop\package\tabletaudiosample.inf.
devcon.exe failed.

The contents of the setupapi.dev.log:
>>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - Root\sysvad_TabletAudioSample]
>>>  Section start 2016/02/03 10:41:27.927
      cmd: devcon.exe  install package\tabletaudiosample.inf Root\sysvad_TabletAudioSample
     dvi: {Update Device Driver - ROOT\MEDIA\0000}
     dvi:      {Build Driver List} 10:41:27.943
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_tabletaudiosample
     dvi:      {Build Driver List - exit(0x00000000)} 10:41:27.959
     dvi:      {DIF_SELECTBESTCOMPATDRV} 10:41:27.959
     dvi:           Using exported function 'MediaClassInstaller' in module 'C:\Windows\system32\mmci.dll'.
     dvi:           Class installer == mmci.dll,MediaClassInstaller
     dvi:           Class installer: Enter 10:41:27.959
     dvi:           Class installer: Exit
     dvi:           Default installer: Enter 10:41:27.975
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed!
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 10:41:27.990
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No drivers were found that are better matching drivers for device instance ROOT\MEDIA\0000 (Error 0x00000103)
!    ndv: No devices were updated.
     dvi: {DIF_DESTROYPRIVATEDATA} 10:41:28.006
     dvi:      Class installer: Enter 10:41:28.006
     dvi:      Class installer: Exit
     dvi:      Default installer: Enter 10:41:28.006
     dvi:      Default installer: Exit
     dvi: {DIF_DESTROYPRIVATEDATA - exit(0xe000020e)} 10:41:28.022
<<<  Section end 2016/02/03 10:41:28.022
<<<  [Exit status: FAILURE(0x00000103)]

Any idea what went wrong? What do I have to do to get this working?

busseklud, 2016-02-28:
I'm getting the same message and have no clue on what to do. Did you figure out how resolve it?

busseklud, 2016-02-28:
There is a bit of help here:

[https://github.com/Microsoft/Windows-driver-samples/issues/21]

I was able to install the driver using the following information:

_tabletaudiosample.inf has a DeviceInstall32 section so you can use "pnputil.exe -a tabletaudiosample.inf" to install it; the physical device object is created by Windows when the driver is added to the store.
C:\Windows\inf\setupapi.dev.log will have more detailed error messages on why the driver is "corrupted or missing"._

After installation I played noise in the right channel and zeros in the left (in Adobe Audition) but the noise is still audible on the right speaker so it seems the swapapo is not in the signal path. Tried to reboot but the issue persists.

Best

alfredo-ossic, 2016-04-03:
"pnputil.exe -a tabletaudiosample.inf" installs the driver, but does not install the swapAPO. Any ideas?

ralusa, 2016-05-11:
I was download the WDK10 and tried to build the sysvad. Initially it shows error such as C2220, C4595. So, I made change in the properties that is "Treat errors as warnings - NO", after this change it successfully builds and generated required dll and cat file. 

Then I have tried to install using "pnputil.exe -a tabletaudisample.inf". Installation success but device is added to "other devices category in device manager ".

pnputil.exe -a tabletaudisample.inf comand prompt logs : 
Processing inf :            tabletaudiosample.inf
Driver package added successfully.
Published name :            oem37.inf
Total attempted:              1
Number successfully imported: 1

setupapi.dev.log
>>>  [SetupCopyOEMInf - C:\install_drivers\tabletaudiosample.inf]
>>>  Section start 2016/05/11 10:42:26.565
      cmd: pnputil.exe  -a tabletaudiosample.inf 
     inf: Copy style: 0x00000000
     sto: {Setup Import Driver Package: C:\install_drivers\tabletaudiosample.inf} 10:42:26.745
     inf:      Provider: TODO-Set-Provider
     inf:      Class GUID: {4d36e96c-e325-11ce-bfc1-08002be10318}
     inf:      Driver Version: 05/10/2016,17.11.14.792
     inf:      Catalog File: sysvad.cat
     sto:      {Copy Driver Package: C:\install_drivers\tabletaudiosample.inf} 10:42:26.763
     sto:           Driver Package = C:\install_drivers\tabletaudiosample.inf
     sto:           Flags          = 0x00000007
     sto:           Destination    = C:\Users\MMTEST~1\AppData\Local\Temp\{97fc79c7-9297-c047-b801-c161ae7bb7f4}
     sto:           Copying driver package files to 'C:\Users\MMTEST~1\AppData\Local\Temp\{97fc79c7-9297-c047-b801-c161ae7bb7f4}'.
     flq:           Copying 'C:\install_drivers\sysvad.cat' to 'C:\Users\MMTEST~1\AppData\Local\Temp\{97fc79c7-9297-c047-b801-c161ae7bb7f4}\sysvad.cat'.
     flq:           Copying 'C:\install_drivers\tabletaudiosample.inf' to 'C:\Users\MMTEST~1\AppData\Local\Temp\{97fc79c7-9297-c047-b801-c161ae7bb7f4}\tabletaudiosample.inf'.
     sto:      {Copy Driver Package: exit(0x00000000)} 10:42:26.798
     pol:      {Driver package policy check} 10:42:26.825
     pol:      {Driver package policy check - exit(0x00000000)} 10:42:26.826
     sto:      {Stage Driver Package: C:\Users\MMTEST~1\AppData\Local\Temp\{97fc79c7-9297-c047-b801-c161ae7bb7f4}\tabletaudiosample.inf} 10:42:26.827
     inf:           {Query Configurability: C:\Users\MMTEST~1\AppData\Local\Temp\{97fc79c7-9297-c047-b801-c161ae7bb7f4}\tabletaudiosample.inf} 10:42:26.838
     inf:                Driver package is fully isolated.
     inf:                Driver package 'tabletaudiosample.inf' is configurable.
     inf:           {Query Configurability: exit(0x00000000)} 10:42:26.842
     flq:           Copying 'C:\Users\MMTEST~1\AppData\Local\Temp\{97fc79c7-9297-c047-b801-c161ae7bb7f4}\sysvad.cat' to 'C:\WINDOWS\System32\DriverStore\Temp\{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\sysvad.cat'.
     flq:           Copying 'C:\Users\MMTEST~1\AppData\Local\Temp\{97fc79c7-9297-c047-b801-c161ae7bb7f4}\tabletaudiosample.inf' to 'C:\WINDOWS\System32\DriverStore\Temp\{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\tabletaudiosample.inf'.
     sto:           {DRIVERSTORE IMPORT VALIDATE} 10:42:26.857
     sig:                {_VERIFY_FILE_SIGNATURE} 10:42:26.872
     sig:                     Key      = tabletaudiosample.inf
     sig:                     FilePath = C:\WINDOWS\System32\DriverStore\Temp\{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\tabletaudiosample.inf
     sig:                     Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\sysvad.cat
!    sig:                     Verifying file against specific (valid) catalog failed! (0x800b0109)
!    sig:                     Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:                {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 10:42:26.878
     sig:                {_VERIFY_FILE_SIGNATURE} 10:42:26.878
     sig:                     Key      = tabletaudiosample.inf
     sig:                     FilePath = C:\WINDOWS\System32\DriverStore\Temp\{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\tabletaudiosample.inf
     sig:                     Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{c848ef65-8e79-5844-b2fc-a8b5d2fca65d}\sysvad.cat
!    sig:                     Verifying file against specific Authenticode(tm) catalog failed! (0x800b0109)
!    sig:                     Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:                {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 10:42:26.882
!    sig:                Driver package catalog file certificate does not belong to Trusted Root Certificates, but user wants to install anyway.
     sto:           {DRIVERSTORE IMPORT VALIDATE: exit(0x00000000)} 10:42:29.041
     sig:           Signer Score = 0x80000000
     sig:           Signer Name  = <unsigned>
     sto:           {DRIVERSTORE IMPORT BEGIN} 10:42:29.043
     bak:                Create system restore point:
     bak:                     Description = Device Driver Package Install: TODO-Set-Provider Sound, video and game controllers
     bak:                     Time        = 0ms
     bak:                     Status      = 0x00000000 (SUCCESS)
     sto:           {DRIVERSTORE IMPORT BEGIN: exit(0x00000000)} 10:42:29.054
[... cut ...]
     dvi:                {Install Root Device: exit(0x00000000)} 10:42:29.248
     sto:           {DRIVERSTORE IMPORT END: exit(0x00000000)} 10:42:29.249
     sto:      {Stage Driver Package: exit(0x00000000)} 10:42:29.250
     sto: {Setup Import Driver Package - exit (0x00000000)} 10:42:29.259
     inf: Driver Store Path: C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_11c7c5be0f5048f7\tabletaudiosample.inf
     inf: Published Inf Path: C:\WINDOWS\INF\oem37.inf
<<<  Section end 2016/05/11 10:42:29.271
<<<  [Exit status: SUCCESS]

girishpattabiraman, 2017-05-02:
Easiest way to install is right-click INF and select install. Using Device manager will result in multiple instances and fail.

closed 2025-11-11 by v-junyli