microsoft/Windows-driver-samples #57: Sysvad fails to install in target PC

alfredo-ossic, 2016-04-04:
Steps:
1-Compile Sysvad using VS2015
2-Provision the driver to the target device using VS2015
3-Run a command window in Administrator mode.
4 cd to the folder where the driver files are located.
5-Run the command: c:\tools\devcon install tabletaudiosample.inf Root\sysvad_TabletAudioSample
Outcome:
---------------------------------------------------------------------------------
Microsoft Windows [Version 10.0.10240]
(c) 2015 Microsoft Corporation. All rights reserved.

C:\Windows\system32>cd c:\Sysvad

c:\Sysvad>c:\tools\devcon install tabletaudiosample.inf Root\sysvad_TabletAudioSample
Device node created. Install is complete when drivers are installed...
Updating drivers for Root\sysvad_TabletAudioSample from c:\Sysvad\tabletaudiosample.inf.
**devcon failed.**

c:\Sysvad>
--------------------------------------------------------------------------------------------------------------------

The setupapi.dev.log:
[Device Install Log]
     OS Version = 10.0.10240
     Service Pack = 0.0
     Suite = 0x0100
     ProductType = 1
     Architecture = x86

[BeginLog]

[Boot Session: 2016/04/04 06:52:04.497]

>>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - Root\sysvad_TabletAudioSample]
>>>  Section start 2016/04/04 07:05:17.523
      cmd: c:\tools\devcon  install tabletaudiosample.inf Root\sysvad_TabletAudioSample
     dvi: {Update Device Driver - ROOT\MEDIA\0000}
     dvi:      Set selected driver complete.
     dvi:      {Build Driver List} 07:05:17.536
     dvi:           Searching for hardware ID(s):
     dvi:                root\sysvad_tabletaudiosample
     dvi:           Processing a single INF: 'c:\sysvad\tabletaudiosample.inf'
     inf:           Opened INF: 'c:\sysvad\tabletaudiosample.inf' ([strings])
     dvi:           Created Driver Node:
     dvi:                HardwareID   - Root\sysvad_TabletAudioSample
     dvi:                InfName      - c:\sysvad\tabletaudiosample.inf
     dvi:                DevDesc      - Virtual Audio Device (WDM) - Tablet Sample
     dvi:                DrvDesc      - Virtual Audio Device (WDM) - Tablet Sample
     dvi:                Provider     - TODO-Set-Provider
     dvi:                Mfg          - TODO-Set-Manufacturer
     dvi:                ModelsSec    - SYSVAD.NTx86
     dvi:                InstallSec   - SYSVAD_SA
     dvi:                Section      - SYSVAD_SA.NT
     dvi:                Rank         - 0x00ff0000
     dvi:                Signer       - WDKTestCert Alfredo,130888044647647603
     dvi:                Signer Score - Authenticode
     dvi:                DrvDate      - 10/15/2015
     dvi:                Version      - 22.46.3.551
     dvi:      {Build Driver List - exit(0x00000000)} 07:05:17.630
     dvi:      {DIF_SELECTBESTCOMPATDRV} 07:05:17.632
     dvi:           Using exported function 'MediaClassInstaller' in module 'C:\Windows\system32\mmci.dll'.
     dvi:           Class installer == mmci.dll,MediaClassInstaller
     dvi:           Using exported function 'ClassCoInstaller_Entry' in module 'C:\Windows\system32\ffusb2audio_coinst.dll'.
     dvi:           CoInstaller 1 == ffusb2audio_coinst.dll,ClassCoInstaller_Entry
     dvi:           CoInstaller 1: Enter 07:05:17.644
     dvi:           CoInstaller 1: Exit
     dvi:           Class installer: Enter 07:05:17.647
     inf:                Opened INF: 'c:\sysvad\tabletaudiosample.inf' ([strings])
     dvi:           Class installer: Exit
     dvi:           Default installer: Enter 07:05:17.656
     dvi:                {Select Best Driver}
     dvi:                     Selected driver installs from section [SYSVAD_SA] in 'c:\sysvad\tabletaudiosample.inf'.
     dvi:                     Class GUID of device remains: {4d36e96c-e325-11ce-bfc1-08002be10318}.
     dvi:                     Set selected driver complete.
     dvi:                     Selected:
     dvi:                          Description - [Virtual Audio Device (WDM) - Tablet Sample]
     dvi:                          InfFile     - [c:\sysvad\tabletaudiosample.inf]
     dvi:                          Section     - [SYSVAD_SA]
     dvi:                          Signer      - [WDKTestCert Alfredo,130888044647647603]
     dvi:                          Rank        - [0x00ff0000]
     dvi:                {Select Best Driver - exit(0x00000000)}
     dvi:           Default installer: Exit
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0x00000000)} 07:05:17.688
     ndv:      Forcing driver install:
     ndv:           Inf Name       - tabletaudiosample.inf
     ndv:           Driver Date    - 10/15/2015
     ndv:           Driver Version - 22.46.3.551
     ndv:      Driver package 'c:\sysvad\tabletaudiosample.inf' exists under 'C:\Windows\System32\DriverStore\FileRepository\tabletaudiosample.inf_x86_fe8ba9a060f09a3c\tabletaudiosample.inf' (oem20.inf).
     dvi:      Searching for hardware ID(s):
     dvi:           root\sysvad_tabletaudiosample
     inf:      Opened PNF: 'C:\Windows\System32\DriverStore\FileRepository\tabletaudiosample.inf_x86_fe8ba9a060f09a3c\tabletaudiosample.inf' ([<no strings>])
!!!  inf:      Could not load INF 'C:\Windows\System32\DriverStore\FileRepository\tabletaudiosample.inf_x86_fe8ba9a060f09a3c\tabletaudiosample.inf'.
!!!  inf:      Error 0xe0000100: The style of the INF is different than what was requested.
     dvi:      Enumerating INFs from path list 'C:\Windows\INF'
     inf:      Searched 8 potential matches in published INF directory
     ndv: {Update Device Driver - exit(e0000203)}
!!!  ndv: Failed to install device instance ROOT\MEDIA\0000 (Error 0xe0000203)
     dvi: {DIF_DESTROYPRIVATEDATA} 07:05:17.726
     dvi:      CoInstaller 1: Enter 07:05:17.728
     dvi:      CoInstaller 1: Exit
     dvi:      Class installer: Enter 07:05:17.732
     dvi:      Class installer: Exit
     dvi:      Default installer: Enter 07:05:17.736
     dvi:      Default installer: Exit
     dvi: {DIF_DESTROYPRIVATEDATA - exit(0xe000020e)} 07:05:17.741
<<<  Section end 2016/04/04 07:05:17.743
<<<  [Exit status: FAILURE(0xe0000203)]

RobKoll, 2016-05-27:
I am fighting the same error code, though the preceding text is slightly different. Log file setupapi.dev.log:
>>>  [Device Install (DiShowUpdateDevice) - ROOT\MEDIA\0000]
>>>  Section start 2016/05/27 14:45:25.722
      cmd: "C:\WINDOWS\system32\mmc.exe" C:\WINDOWS\system32\devmgmt.msc
     dvi: {DIF_UPDATEDRIVER_UI} 14:45:25.724
     dvi:      Using exported function 'MediaClassInstaller' in module 'C:\WINDOWS\system32\mmci.dll'.
     dvi:      Class installer == mmci.dll,MediaClassInstaller
     dvi:      No CoInstallers found
     dvi:      Class installer: Enter 14:45:25.728
     dvi:      Class installer: Exit
     dvi:      Default installer: Enter 14:45:25.728
     dvi:      Default installer: Exit
     dvi: {DIF_UPDATEDRIVER_UI - exit(0xe000020e)} 14:45:25.730
     ndv: {Update Driver Software Wizard for ROOT\MEDIA\0000}
     ndv:      Driver package 'e:\apo\sysvadoriginalbuildresults\tabletaudiosample.inf' exists under 'C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_36f54136dae4e5d9\tabletaudiosample.inf' (oem6.inf).
     dvi:      Searching for hardware ID(s):
     dvi:           root\sysvad_tabletaudiosample
     inf:      Opened PNF: 'C:\WINDOWS\System32\DriverStore\FileRepository\tabletaudiosample.inf_amd64_36f54136dae4e5d9\tabletaudiosample.inf' ([???])
     dvi:      Enumerating INFs from path list 'C:\WINDOWS\INF'
     inf:      Searched 1 potential matches in published INF directory
     ndv: {Update Driver Software Wizard exit(e0000203)}
<<<  Section end 2016/05/27 14:45:49.258
<<<  [Exit status: FAILURE(0xe0000203)]

According to ntstatus.h this may be STATUS_USER_SESSION_DELETED.
But there is a slight mismatch in the error code (0xE0000203 versus 0xC0000203).
Note that this happens with a sysvad build out of the box (selecting debug x64) on a machine which was cleanly installed and only used for experiments with sysvad (tools like notepad++ and fiezilla are installed). Previous to this install of sysvad it has been installed remotely from MSVC 2015 from the development machine (via user wdkremoteuser). This error happens when being logged in as another user (in a domain, development is in smae domain, same user).

Please indicate which additional information may be of help here.

Note also the error code 0xE000020e, STATUS_TOO_MANY_NODES. That may indicate some kind of resource shortage. However, there is no information onto what resources or nodes are at stake here. Is there a limit on device nodes? Would it help to list all devices?

RobKoll, 2016-05-30:
It appears that using
pnputil -i -a tabletaudiosample.inf
will install the sysvad driver.
The driver appears to be working (after reboot).
It will successfully write the audio played into one of the virtual devices to a fie on C:\.

However, some issues remain:
- Upon opening the playback devices (context menu for speaker symbol in systray), the system reports a number of problems with audio enhancements. The system offers to disable thedriver enhancements. 
- After opening the playback devices (context menu for speaker symbol in systray), all playback devices are shown. The properties for e.g. Speakers (virtual audio device (WDM) - Tablet Sample) can be shown, with 3 tabs (General, Levels, Advanced). There is no tab for the specific GUI related to this device. There should be a tab for the GUI as defined in the PropPageExtensions project from the sysvad solution.

PLease inform on how it is possible to install the sysvad driver with complete support for the included SwapApo.

Aurelius1015, 2016-06-12:
Hi RobKoll, I have the same issue for this driver. May I know if you have any feedback from windows or any solution?  If yes, could you please share to me?

RobKoll, 2016-06-13:
Hello Aurelius,

No feedback from Microsoft untill now.However, after discussing the issue with a co-worker at my location in the Netherlands he noted that the driver in itself is installed and working. The SwapApo and the corresponding GUI not. The latter two are .dll. So it looks like a load failure of .dll. Issues regarding loading .dll require investigation with a tool called Dependency Walker (http://www.dependencywalker.com/). What often happens is that a .dll required some other .dll files to be present. If those files are not present, the load of the .dll fails, WITHOUT any hint regarding what went wrong. So please inspect the swapapo.dll and proppageext.dll using dependency walker. Hopefully it reports some missing dependency. That should hint you towards corrective action. In my case it was some .dll which is part of the C run time from MSVC. Probably the MSVC run time was not correctly and completely installed on the host machine. Stupid thing is that I have been bitten by the MissingDllMonster before, and failed to recognize this for now.
Kind regards,
Rob Koll

Aurelius1015, 2016-06-14:
Thanks for your help, it helps me a lot. I utilize Visual C++ Redistributalbe packages to install runtime and GUI is workable. However, I ignored stdunk.h c4945 warning and makes me worrying that there would be unknown issues. In wdmaudiodev, I saw that u have fixed the warning. Could you please educate my how to do it?

RobKoll, 2016-06-14:
Aurelius,

To get rid of the warnings a number of steps need to be done:
- Replace the system stdunk.h header file by a copy, placed in the sysvad root.
- Remove all inline specifiers. All sub projects that use stdunk.h must get an additianl include directive, pointing to the sysvad root (use $(Solutiondir) to avoid absolute paths).
- As the source is no more inline, an additional source file stdunk.c is required. It must have the implementations that you may copy from the original system stdunk.h include file. But avoid the inline specifier.....
- Add the new sourcefile to any project using stdunk.h.
That is the essence of the solution. Tomorrow I will check my new files (stdunk.xxx) and add to this tread (must be sure to remove any hints to the source company and its IP).

Note that the warning has been introduced recently. As far as I can judge, the reason to add it is not an issue in sysvad. Anyway, avoiding the warning is better, Note also that the code will perform slightly worse (added function call overhead added).

Please let me know if further information is required.

Kind regards,
Rob Koll

RobKoll, 2016-06-15:
[stdunk_cpp.txt](https://github.com/Microsoft/Windows-driver-samples/files/315473/stdunk_cpp.txt)
[stdunk_h.txt](https://github.com/Microsoft/Windows-driver-samples/files/315474/stdunk_h.txt)

Feel free to contact me in case more information is needed.

Hi Aurelius,

After some juggling with file names and types it looks like the sources have been uploaded.
Feel free to use them (don't know how MS looks at this, but the root source is public as well).

Kind regards,
Rob Koll

Aurelius1015, 2016-06-17:
Rob Koll:

Thanks for your help. I am wondering if anybody has faced the same issue as I do.
 I have successfully complied and deployed the sysvad project and installed as virtual driver. However, when I set the virtual driver as the default sound device, the produced host stream files have some defects randomly on different computers. The sound defects includes noises and some lags. When we observe the noises in waveform, the noises are in full volume in both sound channels. Interestingly, the noises are randomly appear and randomly disappear in different computers. I found that the noises occurs when choosing default sampling rate, which is 44.1kHz, 16 bit, and there will be no noise when choosing 96kHz, 16bit. In addition, no matter which sampling rate is chosen, the small lags in recording still show up. I wonder if this is an issue from codes or driver itself, thank you! 

Best regards,
Aurelius

RobKoll, 2016-06-18:
Aurelius: I'm glad that you managed to get things running.

Just to add my experience: yes I did observe the noise and silence in the output files (sometimes). I din't pay attention to that. At this point, I need to figure out how to configure a system to perform some testing on an APO I need to write. It needs to contain our own source code (call it legacy code for the purpose of this thread).

Regards,
Rob Koll

neasoon, 2016-08-11:
@RobKoll 

An alternative to fix the inline issue is to suppress the warning C4595 in the affected projects:

Property -> Configuration Properties -> C/C++ -> Advanced -> Disable Specific Warnings

Just add 4595; at the beginning of the line.

I think the affected projects are EndpointsCommon, PhoneAudioSample, and TabletAudioSample.

qwill, 2017-03-31:
Old post but here's my issue:
pnputil -i -a tabletaudiosample.inf
Microsoft PnP Utility

Processing inf :            tabletaudiosample.inf
Adding the driver package failed : The system cannot find the file specified.

Total attempted:              1
Number successfully imported: 0

girishpattabiraman, 2017-05-02:
Installation issues - Easiest way to install is to right-click on the INF and select install. Using device manager will create duplicate device nodes with failing multiple instances.
Audio Enhancements - The APO DLL needs to be signed with the same certificate as installed on the test device, otherwise DLL will fail to load.

AndreiHarasemiuc, 2022-06-23:
Anyone one know why the APO doesn't wind up in the registry in the places it should?
Like HKCR/CLSID  or HKCR/AudioEngine/AudioProcessingObjects?
I successfully install the driver + APO (or so the output tells me) but when checking the registry the APO is not there

closed 2025-11-11 by v-junyli