microsoft/Windows-driver-samples #62: building sysvad for dragon board

bartholmberg, 2016-05-16:
Has anyone built sysvad for a windows IOT yet?   Following the instruction in https://msdn.microsoft.com/en-us/windows/hardware/drivers/develop/getting-started-with-universal-drivers

I am seeing a bunch of missing functions from from API validation.

1>ApiValidation : warning : API CreatePropertySheetPageW in comctl32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API DestroyPropertySheetPage in comctl32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API CheckDlgButton in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API CreateDialogParamW in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API CreateWindowExW in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API DestroyWindow in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API EnableWindow in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API GetClientRect in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API GetDlgCtrlID in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API GetDlgItem in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API GetParent in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API GetWindowLongW in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API IsDlgButtonChecked in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API SendMessageW in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API SetWindowLongW in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API SetWindowPos in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>     1>
1>ApiValidation : warning : API SetWindowTextW in user32.dll is not a supported universal API. PropPageExt.dll calls this API.
1>         ApiValidation: API RegCreateKeyTransactedW in advapi32.dll is not a supported universal API. PropPageExt.dll has a dynamic dependency on this API. If PropPageExt.dll calls the API directly it will not be a universal driver.
1>         ApiValidation: API RegDeleteKeyExW in advapi32.dll is not a supported universal API. PropPageExt.dll has a dynamic dependency on this API. If PropPageExt.dll calls the API directly it will not be a universal driver.
1>         ApiValidation: API RegDeleteKeyTransactedW in advapi32.dll is not a supported universal API. PropPageExt.dll has a dynamic dependency on this API. If PropPageExt.dll calls the API directly it will not be a universal driver.
1>         ApiValidation: API RegOpenKeyTransactedW in advapi32.dll is not a supported universal API. PropPageExt.dll has a dynamic dependency on this API. If PropPageExt.dll calls the API directly it will not be a universal driver.
1>         ApiValidation: API RegQueryValueExW in advapi32.dll is not a supported universal API. PropPageExt.dll has a dynamic dependency on this API. If PropPageExt.dll calls the API directly it will not be a universal driver.
1>         ApiValidation: Driver located at C:\Users\bart.holmberg\Desktop\AT\practice\DriverSamples\audio\sysvad\SwapAPO\PropPageExtensions\ARM\Debug is NOT a Universal Driver
1>     1>
1>C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1747,5): error MSB3721: The command ""C:\Program Files (x86)\Windows Kits\10\bin\x64\ApiValidator.exe" -DriverPackagePath:C:\Users\bart.holmberg\Desktop\AT\practice\DriverSamples\audio\sysvad\SwapAPO\PropPageExtensions\ARM\Debug\ -SupportedApiXmlFiles:"C:\Program Files (x86)\Windows Kits\10\build\universalDDIs\arm\UniversalDDIs.xml" -ModuleWhiteListXmlFiles:"C:\Program Files (x86)\Windows Kits\10\build\universalDDIs\arm\ModuleWhiteList.xml" -ApiExtractorExePath:"C:\Program Files (x86)\Windows Kits\10\bin\x64"" exited with code -1.

girishpattabiraman, 2017-05-02:
Property page DLL is not supported for IOT. The rest of the drivers should build and the INF doesn't actually include the Property page DLL.

bartholmberg, 2020-12-27:
test

closed 2020-12-27 by bartholmberg