microsoft/Windows-driver-samples #64: ndislwf - SDV reports a MustFix Defect (ndisallocatememorywithtagpriority_clean)

mwuetschner, 2016-06-20:
Running the Static Driver Verifier (SDV) for the filter (ndislwf) sample reports a MustFix 
defect ndisallocatememorywithtagpriority.
What is the reason for this.
[image]

jtippet, 2016-06-20:
Thanks for the issue report.  You can consider this non-blocking.  This was a defect in the SDV rule.  We have fixed the rule in our latest builds of SDV.  Which version of SDV are you using?
(For reference, our internal issue number is 6774031).

mwuetschner, 2016-06-21:
I am using the latest WDK 10.0.10586.0.
So I guess Visual Studio 2015 is using the StaticDV.exe from this version which is 4.0.505.8.
Do you have an updated version?

banesttahp1, 2016-07-26:
No I don't, can you please give me instructions on how to fix my macbook in the command Administration X .diskpart.exe,  the partition were windows is stored is locked. I don't know how to fix it.

closed 2018-01-17 by wm1