microsoft/Windows-driver-samples #65: tabletaudiosample audio capture

nbans, 2016-07-09:
Hi,
I have a question about the tabletaudiosample driver. I could build the driver and install it successfully after going through some of the forums. Thanks. I am also able to run the driver and confirm that it saves the speaker output into the file. Next, I am trying to do the loop capture (using wasapi) on this sysvad device but I notice that loop capture, captures the tones instead of the audio that is saved in the file. My expectation was that using WASAPI, I will get the same capture that is stored in the file by the tabletaudiosample.

Can someone let me know that how to disable the tones so that loopback capture can capture the same audio that is saved in the file? I am new to the world of windows drivers and windows OS itself so not much familiar with the architecture. I tried to look at the code but could not figure it out. 

-Thanks

Aurelius1015, 2016-08-08:
Hi nbans , I have the same issue for this driver. May I know if you have any feedback from windows or any solution? If yes, could you please share to me?

redwhale13, 2016-08-08:
They want me to kill the thread or keep it

redwhale13, 2016-08-17:
Its mine jrb

nbans, 2016-08-17:
Hi Aurelius1015, Sorry for not replying early. I got the feedback from Microsoft that loopback capture is not supported on the sysvad. So currently there is no solution as I know. To have support of loopback capture on the sysvad, I was told that one has to write mixer. I didn't spend further effort on it.

closed 2025-11-11 by v-junyli