microsoft/Windows-driver-samples #89: cannot open port settings of HyperTerminal Using WDFSerial driver sample under windows xp

Mapleaph, 2016-11-07:
I am new to windows device driver development, I have a PCI bus device and I want to test the windows wdfserial.sys device driver on the device under windows XP, the communication is OK, but I just find that, when I'm using the system supplied HyperTerminal.exe to set the properties of a COM port, there is no pop-up window of port settings tab when I click the settings button, Any idea?

AlexeiFedorov, 2016-11-07:
Does “mode COMx” command work for you?

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

Mapleaph, 2016-11-14:
Thanks for your reply @AlexeiFedorov , the command you formerly mentioned does work for me, but I think I found where the problem is. I just pulled a request to submit my modification of the source code.

closed 2017-03-15 by wm1

wm1, 2017-03-16:
Merge pull request #93 - Fix NULL-terminating device name when writing to DEVICEMAP/SERIALCOMM