microsoft/Windows-driver-samples #91: Seems no .vcxproj files in Windows-driver-samples-master\audio\sysvad

danteliujie, 2016-11-09:
Hi Team,

Thank for your great work.
I downloaded the zip file and found there are no .vcxproj in the folder Windows-driver-samples-master\audio\sysvad, 
And checked on github.com, didn't found the files.
Have I missed something? Could someone help to give me some suggestions or fix it?
If there are forum, please point me to that.

Thanks

fsheikh, 2016-11-20:
I encountered this as well, could someone please respond?

5783mgordon, 2016-11-22:
Same here... any help?

fsheikh, 2016-11-23:
For the benefit of others I was able to build the virtual audio driver by using an older commit:
https://github.com/Microsoft/Windows-driver-samples/commit/96eb96dfb613e4c745db6bd1f53a92fe7e2290fc

Like this: https://github.com/Microsoft/Windows-driver-samples/blob/687b274aa38fd05c8c26e3068932121876d7f745/audio/sysvad/EndpointsCommon/EndpointsCommon.vcxproj

I then had to change project build settings to stop treating warnings as errors and manually added `stdafxsrc.cpp`

With these changes I was able to build sysvad project/solution. But it does look like these windows 10 driver samples have bit rotten a bit.

wm1, 2017-03-15:
Fixed recently

closed 2017-03-15 by wm1