microsoft/Windows-driver-samples #101: Vendor name string for 0x096E is wrong

Geoffrey-Li, 2016-12-26:
In file https://github.com/Microsoft/Windows-driver-samples/blob/master/usb/usbview/vndrlist.h, there is a vendor list for USB devices, but the name for 0x096E item, it should be "Feitian Technologies Co., Ltd." From http://www.the-sz.com/products/usbid/index.php?v=096E&p=&n= , you can see the Manufacture website is www.ftsafe.com, so you can find the name from the website.

MichelleBergeron, 2017-01-27:
Thanks - we're tracking this on our end and we'll fix it in a future version.

- Michelle B. @ Windows USB team

wm1, 2017-03-15:
The data comes from USB Implementers Forum (usb.org) directly. Microsoft does not maintain a separated list.

The real solution is for the vendor to contact usb.org and have their registered name updated. Microsoft will then pull the vendor list from there.

closed 2017-03-15 by wm1

Geoffrey-Li, 2017-03-16:
I can not get this information directly from usb.org, but from http://www.linux-usb.org/usb.ids our company is named "Feitian Technologies, Inc.". I think this is ok for us, "Feitian **New** Technology Co."  in https://github.com/Microsoft/Windows-driver-samples/blob/master/usb/usbview/vndrlist.h is totally wrong.

wm1, 2017-03-16:
The USB Implementers Forum is the authority which assigns and maintains all USB Vendor ID Numbers. Check out "Getting a Vendor ID" page at http://www.usb.org/developers/vendor/ for start. Further inquiry on how to update usb.org's records on company name should also be directed to usb.org as well.

Geoffrey-Li, 2017-03-16:
I have sent email to admin@usb.org, hope get the response ASAP.
Thank you very much.

Geoffrey-Li, 2017-03-31:
usb.org has corrected our company name, please update the list.

wm1, 2017-04-03:
The list is not updated on their website yet. I'll check it again in one month. You can ping me here if I forget to do so.

Geoffrey-Li, 2017-04-17:
I have got response from usb.org, they update this list quarterly, last time is March 2, so  the list maybe will be updated next time at June 2017.

Geoffrey-Li, 2017-07-17:
@wm1 ,
Would you please help to check whether usb.org has updated the corresponding company name to 0x096E now?

wm1, 2017-07-17:
vndrlist.h is updated. Please check.

Geoffrey-Li, 2017-07-18:
thank you very much.