microsoft/Windows-driver-samples #107: issue with AvsCamera?

andeb91, 2017-01-10:
One month ago I've tried this driver on my pc doing a feasibility study for a project and everithing was going well. Today I've started this project but now if I test the driver, Windows give me the blue screen and automatically restarts.  
Could someone help me?

zodiacon, 2017-01-10:
check out the dump file in c:\windows\memory.dmp - open in WinDbg - you should get some hints

andeb91, 2017-01-11:
[MemoryDump.txt](https://github.com/Microsoft/Windows-driver-samples/files/698177/MemoryDump.txt)

Here the output of WinDbg

zodiacon, 2017-01-11:
issue !analyze -v and observe the call stack...

andeb91, 2017-01-11:
Thank you for helping me, but this is my first time that i write a kernel-mode driver.
I'm runnung the AvsCamera sample without changing nothing.
Here the result of your hint

[MemoryDump.txt](https://github.com/Microsoft/Windows-driver-samples/files/698207/MemoryDump.txt)

zodiacon, 2017-01-11:
It's not easy remote debugging like this... you need the symbols for AvsCamera.sys to see where things fail. Choose File->Symbol File Path from WinDbg menu and add the folder where AvsCamera.pdb is at with a semicolon, something like "SRV*c:\....*http://msdl...;c:\myfolderWherePdbIs
and then check the box reload. Then issue the k command - you should see the function name that faulted.

andeb91, 2017-01-11:
I've added in the Symbol File Path the folder where AvsCamera.pdb is but it still not working, it say "ERROR: Module load completed but symbols could not be loaded for AvsCamera.sys"

zodiacon, 2017-01-11:
try .reload /f avscamera.sys
if this fails, then for some reason it's probably the wrong pdb
You can also try !sym noisy to get more information when loading symbols

andeb91, 2017-01-11:
It seems that it try to load the pdb of an old example modified by me one month ago.
I run  !chksym  AvsCamera and the pdb path is the old example pdb that I've deleted immediately. 
How can I remove it?

ntwerdochlib, 2017-01-12:
sounds like your symbol path is wrong?  what does the output of .sympath show

andeb91, 2017-01-12:
the sympath is ok, it contains the path to my new pdb file, but the output of "!chksym AvsCamera" show a different path for the pdb

yupi2, 2017-01-12:
Rebuild so it'll use a new pdb then blue-screen with the rebuilt version and grab the dump again.

andeb91, 2017-01-12:
Unfortunally I've already tryed it a lot of time yesterday, but nothing seems change.

The AvsCamera.sys file remain in C:\Windows\System32\drivers folder also if I 'uninstall' the driver. Maybe is this the problem?

ntwerdochlib, 2017-01-12:
First, I know issues like this are a real pain; been there....Here is what I do:Add a DbgPrint statement in DriverEntry that prints the values of __DATE__ and __TIME__ (you might have to enable date time support in the project config)Do a full rebuild of the driverOn the target system:uninstall the driveruser devcon dp_enum to list all driver packagesif your driver package shows up, use devcon dp_delete oem[#].inf to delete itgo to %systemroot%\system32 and do a dir /s [driver file name]* to see if anything is left behind, if so remove itlastly I do a sc query [driver service name] just to see if the service is still listed.  sometimes it will still show as STOPPED.  use sc delete [dririver service name] to force its removaldouble check the sympath in windbg; it helps to keep this a simple as possible.once the sympath is good, I will add the ms symbol server using .symfix+when the driver loads, your debug line should show with acceptable values for date time, so you are confident the most current driver is running.If I have time today I will try building the avscamera example and try to load it to see if there is anything I run into.

closed 2017-03-18 by wm1