microsoft/Windows-driver-samples #109: No kbfiltr.inf file

JPG-Consulting, 2017-01-19:
When I compile the kbdfilter project I get no .inf file and therefore installation steps fail. How do you create the .inf file?

fischmax, 2017-02-25:
I have the exact same issue when trying to build with msbuild. Renaming the kbfiltr.inx to kbfiltr.inf makes stampinf run but I am not too sure what is going on there (and I can't run the resulting driver either). I haven't tried building with Visual Studio yet, since I kinda don't want to install it ;)

Also, readme.md speaks of a `[DDK_Ex.Mfg.NT$ARCH$]` section in the inx file which appears to be missing …

jcclark53, 2017-02-26:
Same issue here, only sys file produced, missing files as in readme:
Kmdfsamples.cat
kbfiltr.inf
WdfCoinstaller010xx.dll

ZloiBubr, 2017-02-26:
Same issue here, only sys file produced, missing files as in readme:
Kmdfsamples.cat
kbfiltr.inf
WdfCoinstaller010xx.dll

fischmax, 2017-03-01:
Only slightly related but I need to vent: following these [instructions](https://msdn.microsoft.com/en-us/windows/hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1) bricked my target computer. It now just BSODs when booting. This is bad because I kinda need that machine :-(

ZloiBubr, 2017-03-01:
@fischmax try F8 key and "Last known good configuration"

jcclark53, 2017-03-02:
😞

rsachira, 2017-03-10:
Add the kbfiltr.inx to the project (Add->Add Existing) then the inf gets generated

closed 2017-03-15 by wm1