microsoft/Windows-driver-samples #113: PCIDRV example does not compile due to WPP errors in pcidrv.c

l0ud, 2017-01-27:
Hi. I had problems with compilation of pcidrv example, inside general/pcidrv directory. 

With compiling for windows 10, using MSVC2015, pcidrv.c file gives the following compilation errors:
`2>..\pcidrv.c(123): error C4013: 'WPP_CALL_pcidrv_c123' undefined; assuming extern returning int
2>..\pcidrv.c(123): error C2065: 'DBG_INIT': undeclared identifier
2>..\pcidrv.c(148): error C4013: 'WPP_CALL_pcidrv_c148' undefined; assuming extern returning int
2>..\pcidrv.c(148): error C2065: 'DBG_INIT': undeclared identifier
2>..\pcidrv.c(176): error C4013: 'WPP_CALL_pcidrv_c176' undefined; assuming extern returning int
2>..\pcidrv.c(176): error C2065: 'DBG_INIT': undeclared identifier
2>..\pcidrv.c(217): error C4013: 'WPP_CALL_pcidrv_c217' undefined; assuming extern returning int
2>..\pcidrv.c(217): error C2065: 'DBG_PNP': undeclared identifier
2>..\pcidrv.c(357): error C4013: 'WPP_CALL_pcidrv_c357' undefined; assuming extern returning int
2>..\pcidrv.c(357): error C2065: 'DBG_PNP': undeclared identifier
2>..\pcidrv.c(369): error C4013: 'WPP_CALL_pcidrv_c369' undefined; assuming extern returning int
2>..\pcidrv.c(369): error C2065: 'DBG_PNP': undeclared identifier
2>..\pcidrv.c(381): error C4013: 'WPP_CALL_pcidrv_c381' undefined; assuming extern returning int
2>..\pcidrv.c(381): error C2065: 'DBG_PNP': undeclared identifier
2>..\pcidrv.c(392): error C4013: 'WPP_CALL_pcidrv_c392' undefined; assuming extern returning int
2>..\pcidrv.c(392): error C2065: 'DBG_PNP': undeclared identifier
2>..\pcidrv.c(396): error C4013: 'WPP_CALL_pcidrv_c396' undefined; assuming extern returning int
2>..\pcidrv.c(396): error C2065: 'DBG_PNP': undeclared identifier
2>..\pcidrv.c(413): error C4013: 'WPP_CALL_pcidrv_c413' undefined; assuming extern returning int
2>..\pcidrv.c(413): error C2065: 'DBG_PNP': undeclared identifier
2>..\pcidrv.c(420): error C4013: 'WPP_CALL_pcidrv_c420' undefined; assuming extern returning int
...`

I figured out that all line numbers in tmh files are off by 1. Problem was fixed by removing extra empty line on the very beginning of pcidrv.c. But why WPP parser ignores that line in the first place?

wm1, 2017-03-15:
No repro

closed 2017-03-15 by wm1

JPBlanc, 2017-07-23:
Thanks for your post, it save me time.