microsoft/Windows-driver-samples #123: Signability test failed.

yjdwbj, 2017-03-18:
hi ,all.
when i build this **https://github.com/Microsoft/Windows-driver-samples/tree/master/print/XPSDrvSmpl** this project occur this error like below:
```
Inf2Cat Tool Output:
.............................
Signability test failed.

Errors:
22.9.1: xdwscrgb.icc in [colorprofiles] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdcmykprinter.icc in [colorprofiles] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl.ini in [configplugin] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdnames.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdwmark.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdbook.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdcolman.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdnup.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdpgscl.gpd in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl-pipelineconfig.xml in [xpsdrvsample] of \xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.

Warnings:
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntx86] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntia64] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntamd64] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntx86.6.0] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntia64.6.0] of \xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntamd64.6.0] of \xdsmpl.inf
```

oneselfz, 2017-05-15:
same problem.

bjm3819, 2017-10-12:
has this been resolved? I'm also having this issue.

stephenegriffin, 2017-12-22:
No one can figure this out:
https://social.msdn.microsoft.com/Forums/sqlserver/en-US/2b7cff68-3769-42b2-bfb0-6b40e046ee2b/inf2cat-tool-signability-test-failed?forum=wdk
https://www.osronline.com/ShowThread.cfm?link=271547
The guy from the last thread seems to understand what to do but doesn't explain it for the rest of the class.

jnrbo, 2018-02-07:
Go to **package** Project and add a reference to the just created Project (Driver Install Package - the one that is referenced at step 3).

closed 2018-07-19 by NeoAdonis

has-syed, 2023-06-30:
My experience has been that if the inf file name, the cat file name or the sys file name don't match, there can be issues.

PR #208 (2018-07-19): Update README.md
Update build instructions. Resolves #123. Resolves #197. Resolves #206. Resolves #31.
commit c97def4e (2018-07-19): Update README.md

Update build instructions. Resolves #123. Resolves #197. Resolves #206. Resolves #31.