microsoft/Windows-driver-samples #130: Keyboard layout driver samples do not compile.

4nd1m4n, 2017-04-18:
Using the explanations i found on the internet i installed Visual Studio 2015 Community, the WinWDK and WinSDK. I also downloaded the keyboard layout samples:

https://github.com/Microsoft/Windows-driver-samples/tree/master/input/layout/kbdus
https://github.com/Microsoft/Windows-driver-samples/tree/master/input/layout/all_kbds/kbdfr
https://github.com/Microsoft/Windows-driver-samples/tree/master/input/layout/all_kbds/kbdgr

which uppon compiling with the Developer Command Line expression: `msbuild /t:clean /t:build .\kbdus.vcxproj` does not compile but rather gives me the error code: 

`c:\Users\USER\Downloads\kbd\kbdus>msbuild /t:clean /t:build .\kbdus.vcxproj
Microsoft (R)-Buildmodul, Version 14.0.25420.1
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

Der Buildvorgang wurde am 18.04.2017 03:49:22 gestartet.
Projekt "c:\Users\USER\Downloads\kbd\kbdus\kbdus.vcxproj" auf Knoten "1", clean;build Ziel(e).
c:\Users\USER\Downloads\kbd\kbdus\kbdus.vcxproj(35,91): error MSB4025: Die Projektdatei konnte nicht geladen werden.
 Unerwartetes Token '>'. Erwartet wurde das Token '='. Zeile 35, Position 91.
Die Erstellung des Projekts "c:\Users\USER\Downloads\kbd\kbdus\kbdus.vcxproj" ist abgeschlossen, clean;build Ziel(e)
 -- FEHLER.

Fehler beim Buildvorgang.

"c:\Users\USER\Downloads\kbd\kbdus\kbdus.vcxproj" (clean;build Ziel) (1) ->
  c:\Users\USER\Downloads\kbd\kbdus\kbdus.vcxproj(35,91): error MSB4025: Die Projektdatei konnte nicht geladen werde
n. Unerwartetes Token '>'. Erwartet wurde das Token '='. Zeile 35, Position 91.

    0 Warnung(en)
    1 Fehler`

As you can see i'm using this verison of the build  program: Microsoft (R)-Buildmodul, Version 14.0.25420.1

So what is wrong?

wm1, 2018-01-16:
kbd\kbdus\kbdus.vcxproj(35,91): error MSB4025: The project file could not be loaded. Unexpected token '>'. The token was expected to be '='. Line 35, position 91.

wm1, 2018-01-16:
The issue does not reproduce on my side. I was using VS 2017 on English system though.

closed 2018-01-16 by wm1