microsoft/Windows-driver-samples #141: Issue with WDK 10.0.15063.0 and umdf2_fx2 driver sample

Tam66662, 2017-05-23:
I'm unable to get the usb/umdf2_fx2.sln to compile with WDK 10.0.15063.0.  I have a completely clean and newly built Virtual Machine with these installations: Windows 10 Anniversary Update (1607) with VS2015 Update 3 installed, and WDK 10.0.15063.0.

Result: With this setup, I can't get the umdf2_fx2.sln to compile at all.  It's returning numerous errors due to missing header files dependencies for 15063.  For example:

Severity	Code	Description	Project	File	Line	Suppression State
Error (active)		cannot open source file "driverspecs.h"	osrusbfx2um	c:\Program Files (x86)\Windows Kits\10\Include\wdf\umdf\2.15\wdf.h	37	

I then tried to "repair" the SDK by doing a Repair from "Programs and Features".  This eliminated some of the above errors, but still some remain.  For example:

Severity	Code	Description	Project	File	Line	Suppression State
Error	C1083	Cannot open include file: 'C:\Program Files (x86)\Windows Kits\10\Include\10.0.15063.0\shared\warning.h': No such file or directory	

These missing header files previously existed in WDK 10.0.14393.0, but not in 10.0.15063.0.  It seems to me as if the WDK 10.0.15063.0 installer is not installing all of the files that I need.  Specifically everything that is in the "shared" folder.  

I don't have a workaround where I can use WDK 10.0.14393.0 because I need the DEVPKEY_DeviceInterface_UnrestrictedCapabilities to be defined in the devpkey.h, which doesn't exist in the header file for 14393.  Online I've researched that this key was added to the [devpkey.h header file as of 15021](https://naughter.wordpress.com/2017/02/14/changes-in-the-windows-v10-0-15021-sdk-compared-to-windows-v10-0-14393-sdk-part-one/).

Can someone help verify that WDK 10.0.15063.0 is missing necessary header files in its installation?  And can someone help me figure out how to fix this?

Thanks,
Tam

FYI, there are 294 files in the WDK 10.0.14393.0 "shared" folder vs only 19 files in the WDK 10.0.15063.0 shared folder after a Repair, and I am in need of a number of the missing header files that existed in the previous 14393 WDK.

Name                           
----                           
acpiioct.h                     
d3dkmddi.h                     
d3dkmdt.h                      
ksarm.h                        
kxarm.h                        
kxarmunw.h                     
nvme.h                         
pointofservicecommontypes.h    
pointofservicedriverinterface.h
smcdbg.h                       
trustedrt.h                    
WpAccelerometer.h              
WpAmbient.h                    
WpFusion.h                     
WpGyro.h                       
WpMagnetometer.h               
WpOrientation.h                
WpProximity.h                  
WpSensors.h

ofirc, 2017-06-07:
Beware: the Windows 10 SDK 10.0.15063.400 **breaks all CMake builds with VS2015**!

I had a similar story. Like @Tam66662 I installed the WDK 10.0.15063.0, but I couldn't build my driver due to missing headers, one of them from shared\warning.h.

After installing the corresponding Win10 SDK 10.0.15063.400, the driver build succeeded but any C++ project via CMake would fail miserably:
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown

The solution was to install a previous version of the WDK and SDK:
WDK 10.0.14393.0
Windows SDK 10.0.14393.795

and now I can build both C++ user-space code and my driver.

Probably worth filing a bug to Microsoft.

rodwiddowson, 2017-06-07:
You cannot build kernel code against the latest sdk. The WDK  hasn't been released for it. So "the software is functioning as written"..,

You need (on vs) to explicitly target the 1.14xyz sdk and not "latest sdk". It's in the project page somewhere. No idea how to do that for CMAKE

ofirc, 2017-06-07:
What is the point in uploading a WDK that is incompatible with the latest SDK?
Or uploading a WDK that requires manual hacks to get it working?

This is not the first time I installed the WDK+SDK, this is, however, the first time that it broke any C/C++ application via latest CMake.

Tam66662, 2017-06-13:
@rodwiddowson, were you directing your comment to me or @ofirc ?  This is not a kernel mode driver sample that I am trying to run...it's a umdf driver.  And I cannot explicitly target the 14393 version due to the reasons that I stated in my original message: 

> I don't have a workaround where I can use WDK 10.0.14393.0 because I need the DEVPKEY_DeviceInterface_UnrestrictedCapabilities to be defined in the devpkey.h, which doesn't exist in the header file for 14393. Online I've researched that this key was added to the devpkey.h header file as of 15021.

douglascghext, 2017-06-30:
Uninstalling and reinstalling WDK and SDK combo fixed the problem for me. For what it is worth MSFT did say some files would get removed and offered a workaround in the WDK page https://developer.microsoft.com/en-us/windows/hardware/windows-driver-kit. I tried the modify and repair approach but that did not work for me even after a reboot.

closed 2017-09-27 by wm1