microsoft/Windows-driver-samples #144: FWPM_LAYER_ALE_CONNECT_REDIRECT_Vx: Can't redirect udp traffic to local process

xjdrew, 2017-06-01:
I set a filter as below:
```
C:\test>WFPSampler.Exe -s PROXY -l FWPM_LAYER_ALE_CONNECT_REDIRECT_V4 -iprp 80 -pra 127.0.0.1 -prp 4080 -v -plspid 1
INFO:  ProxyScenarioExecute : RpcInvokeScenarioProxy() [status: 0]
```
the filter will match any traffic whose remote port is 80 and redirect it 127.0.0.1:80. I test with TCP protocol, it works perfectly:

```
# listen on tcp 4080
nc -L -p 4080

# connet to any address's tcp 80 port
nc -n -v 192.168.56.1 80
```

But when I test with udp traffic, It fails.
```
#server: listen on udp 4080
nc -L -u -p 4080

#client: connet to any address's udp 80 port
nc -n -u -v 192.168.56.1 80
```
I send some packet from the client, the server can't receive any packets. And the client also don't receive any error so the flow keeps all the time.

WPP trace shows the flow is redirected by the callout:

> 00000105	WFPSamplerCallouts	5348	3544	0	105	05\31\2017-19:27:22:623	 ---> ClassifyProxyByALERedirect() [Layer: FWPS_LAYER_ALE_CONNECT_REDIRECT_V4][FilterID: 0x1104d][Rights: 0x1]
00000106	WFPSamplerCallouts	5348	3544	0	106	05\31\2017-19:27:22:623	 ---> KrnlHlprRedirectDataCreate()
00000107	WFPSamplerCallouts	5348	3544	0	107	05\31\2017-19:27:22:623	 ---> KrnlHlprRedirectDataPopulate()
00000108	WFPSamplerCallouts	5348	3544	0	108	05\31\2017-19:27:22:623	 <--- KrnlHlprRedirectDataPopulate() [status: 0]
00000109	WFPSamplerCallouts	5348	3544	0	109	05\31\2017-19:27:22:623	 <--- KrnlHlprRedirectDataCreate() [status: 0]
00000110	WFPSamplerCallouts	5348	3544	0	110	05\31\2017-19:27:22:623	 ---> TriggerProxyByALERedirectOutOfBand()
00000111	WFPSamplerCallouts	5348	3544	0	111	05\31\2017-19:27:22:623	 ---> KrnlHlprClassifyDataCreateLocalCopy()
00000112	WFPSamplerCallouts	5348	3544	0	112	05\31\2017-19:27:22:623	 ---> KrnlHlprClassifyDataAcquireLocalCopy()
00000113	WFPSamplerCallouts	5348	3544	0	113	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpsIncomingValuesCreateLocalCopy()
00000114	WFPSamplerCallouts	5348	3544	0	114	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpsIncomingValuesPopulateLocalCopy()
00000115	WFPSamplerCallouts	5348	3544	0	115	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000116	WFPSamplerCallouts	5348	3544	0	116	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpByteBlobCreateLocalCopy()
00000117	WFPSamplerCallouts	5348	3544	0	117	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpByteBlobPopulateLocalCopy()
00000118	WFPSamplerCallouts	5348	3544	0	118	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpByteBlobPopulateLocalCopy() [status: 0]
00000119	WFPSamplerCallouts	5348	3544	0	119	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpByteBlobCreateLocalCopy() [pByteBlob: FFFFCE8C73499090]
00000120	WFPSamplerCallouts	5348	3544	0	120	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000121	WFPSamplerCallouts	5348	3544	0	121	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000122	WFPSamplerCallouts	5348	3544	0	122	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpByteBlobCreateLocalCopy()
00000123	WFPSamplerCallouts	5348	3544	0	123	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpByteBlobPopulateLocalCopy()
00000124	WFPSamplerCallouts	5348	3544	0	124	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpByteBlobPopulateLocalCopy() [status: 0]
00000125	WFPSamplerCallouts	5348	3544	0	125	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpByteBlobCreateLocalCopy() [pByteBlob: FFFFCE8C73388440]
00000126	WFPSamplerCallouts	5348	3544	0	126	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000127	WFPSamplerCallouts	5348	3544	0	127	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000128	WFPSamplerCallouts	5348	3544	0	128	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000129	WFPSamplerCallouts	5348	3544	0	129	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000130	WFPSamplerCallouts	5348	3544	0	130	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000131	WFPSamplerCallouts	5348	3544	0	131	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000132	WFPSamplerCallouts	5348	3544	0	132	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000133	WFPSamplerCallouts	5348	3544	0	133	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000134	WFPSamplerCallouts	5348	3544	0	134	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000135	WFPSamplerCallouts	5348	3544	0	135	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000136	WFPSamplerCallouts	5348	3544	0	136	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000137	WFPSamplerCallouts	5348	3544	0	137	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000138	WFPSamplerCallouts	5348	3544	0	138	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000139	WFPSamplerCallouts	5348	3544	0	139	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000140	WFPSamplerCallouts	5348	3544	0	140	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000141	WFPSamplerCallouts	5348	3544	0	141	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000142	WFPSamplerCallouts	5348	3544	0	142	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000143	WFPSamplerCallouts	5348	3544	0	143	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000144	WFPSamplerCallouts	5348	3544	0	144	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000145	WFPSamplerCallouts	5348	3544	0	145	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
00000146	WFPSamplerCallouts	5348	3544	0	146	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValuePopulateLocalCopy() [status: 0]
00000147	WFPSamplerCallouts	5348	3544	0	147	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValuePopulateLocalCopy()
[... 43 lines cut ...]
00000191	WFPSamplerCallouts	5348	3544	0	191	05\31\2017-19:27:22:623	 <--- KrnlHlprDPCDataCreate() [status: 0]
00000192	WFPSamplerCallouts	5348	3544	0	192	05\31\2017-19:27:22:623	 ---> ProxyByALERedirectDeferredProcedureCall()
00000193	WFPSamplerCallouts	5348	3544	0	193	05\31\2017-19:27:22:623	 ---> PerformProxyConnectRedirection()
00000194	WFPSamplerCallouts	5348	3544	0	194	05\31\2017-19:27:22:623	 ---> KrnlHlprFwpValueGetFromFwpsIncomingValues()
00000195	WFPSamplerCallouts	5348	3544	0	195	05\31\2017-19:27:22:623	 <--- KrnlHlprFwpValueGetFromFwpsIncomingValues() [pValue: FFFFCE8C70D89060]
00000196	WFPSamplerCallouts	5348	3544	0	196	05\31\2017-19:27:22:623	 ---> KrnlHlprRedirectDataDestroy()
00000197	WFPSamplerCallouts	5348	3544	0	197	05\31\2017-19:27:22:623	 ---> KrnlHlprRedirectDataPurge()
00000198	WFPSamplerCallouts	5348	3544	0	198	05\31\2017-19:27:22:623	 <--- KrnlHlprRedirectDataPurge()
00000199	WFPSamplerCallouts	5348	3544	0	199	05\31\2017-19:27:22:623	 <--- KrnlHlprRedirectDataDestroy()
00000200	WFPSamplerCallouts	5348	3544	0	200	05\31\2017-19:27:22:623	 <--- PerformProxyConnectRedirection() [status:0]
00000201	WFPSamplerCallouts	5348	3544	0	201	05\31\2017-19:27:22:623	 ---> KrnlHlprDPCDataDestroy()
00000202	WFPSamplerCallouts	5348	3544	0	202	05\31\2017-19:27:22:623	 ---> KrnlHlprDPCDataPurge()
00000203	WFPSamplerCallouts	5348	3544	0	203	05\31\2017-19:27:22:623	 <--- KrnlHlprDPCDataPurge()
00000204	WFPSamplerCallouts	5348	3544	0	204	05\31\2017-19:27:22:623	 <--- KrnlHlprDPCDataDestroy()
00000205	WFPSamplerCallouts	5348	3544	0	205	05\31\2017-19:27:22:623	 <--- ProxyByALERedirectDeferredProcedureCall()
00000206	WFPSamplerCallouts	5348	3544	0	206	05\31\2017-19:27:22:623	 <--- KrnlHlprDPCQueue() [status: 0]
00000207	WFPSamplerCallouts	5348	3544	0	207	05\31\2017-19:27:22:623	 <--- TriggerProxyByALERedirectOutOfBand() [status: 0]
00000208	WFPSamplerCallouts	5348	3544	0	208	05\31\2017-19:27:22:623	 <--- ClassifyProxyByALERedirect() [Layer: FWPS_LAYER_ALE_CONNECT_REDIRECT_V4][FilterID: 0x1104d][Action: 0x1001][Rights: 0x1][Absorb: FALSE]

I also try redirect udp traffic to remote machine, it works fine.

I think ``` Can't redirect udp traffic to local process``` is a bug of WFP or WFPSampler.

More information(I have asked in MSDN's WFP Forums, none answer):
https://social.msdn.microsoft.com/Forums/sqlserver/en-US/b57f1aef-5054-4d21-8166-cb3690f32562/fwpslayeraleconnectredirectv4-redirect-udp-traffic-to-local-process-doesnt-work?forum=wfp

xjdrew, 2017-06-01:
@jmrossy @girishpattabiraman @wm1 

please advice.

GreatFireWall, 2017-12-26:
@xjdrew 请问你怎样成功编译运行的? 解决方案里根本没有文档所说的 package 项目. 我自建了一个之后, 在虚拟机搭建好环境, 运行 WFPSamplerInstall.cmd 又提示  WFPSamperServer 无法运行. 还有我看了 WFPSamplerInstall.cmd 里面的实现, 使用 signtool sign -A -V WFPSamepler.exe 进行签名会提示 "SignTool Error: No certificates were found that met all the given criteria." .  请问都是怎么解决的?

xjdrew, 2017-12-27:
vs2017配置好驱动开发环境，可以直接编译。

GreatFireWall, 2017-12-27:
编译是可以的, 但是部署的时候不能运行, 那个部署脚本 WFPSamplerInstall.cmd 直接在 Target Compluter 运行也有点问题.

xjdrew, 2017-12-28:
打开Windows的内核调试模式没有呢？

closed 2025-11-11 by v-junyli