microsoft/Windows-driver-samples #154: Virtual HID Framework (VHF): need xample

TronicLabs, 2017-07-27:
We need some example.

FansteKaRiNa1998, 2017-08-24:
There is one, but I can't get it to working, it's like I am misconfiguring lower filter to that driver or something, all hid reports get STATUS_SUCCESS but nothing changing in the USB Game Controllers
https://github.com/ms-iot/samples/tree/develop/HIDInjector

BestBurning, 2020-03-16:
We need some example.

Surferlul, 2022-10-18:
A more current version of HIDInjector is available here: https://github.com/microsoft/Windows-IoT-Samples/tree/master/samples/HIDInjector

daemonfire300, 2025-03-16:
I got this one to compile, when I run it on my Windows 11 VM, everything returns a successful NTSTATUS or otherwise non-error returns, but: There is not a single keystroke being sent to Notepad.exe for example, even though the Window has focus.
I checked the Device Manager, the Virtual Device is there including the VHF Device including all the other virtual devices (mouse, keyboard etc).

yangtb2, 2025-08-26:
need sample with UMDF

v-junyli, 2025-11-11:
@microsoft/coreos-buses-and-wdf customer still have request for this sample. Assigning it to your team, thank you.