microsoft/Windows-driver-samples #172: The Readme.MD file for master/audio/sysvad needs to be corrected.

levmin, 2017-10-10:
In the Build and Deployment sections, there are several omissions and typos.

To build the sample, at this time one needs Visual Studio 2015. VS2017 is not supported as of yet.

For manual deployment, an attempt to run a command bcdedit /set TESTSIGNING ON may ( in fact, likely will)  produce an error message: 

An error has occurred setting the element data.
The value is protected by Secure Boot policy and cannot be modified or deleted.

One should go into UEFI and disable Secure Boot there. UEFI can be accessed via Start > Settings > Update & Security > Recovery > Advanced startup > Restart now > Troubleshoot > Advanced options > UEFI Firmware Settings > Restart.

In order to properly import the driver certificate, it is essential in the Certificate Import Wizard to select "Local Machine" and then place the certificate into Trusted Root Certification Authority, otherwise the driver will fail to install.

The sample drivers are unsigned. Windows 10 x64 will not load them unless it is started with Driver Signature Enforcement disabled. Start > Settings > Update & Security > Recovery > Advanced startup > Restart now >Troubleshoot > Advanced options > Startup Settings > 7 or F7.

The correct command line to install the driver via devcon is:

devcon install tabletaudiosample.inf ROOT\sysvad_TabletAudioSample

gooderist, 2018-12-21:
I was able to build with VS2017 (after a bit of work). The devcon install instructions definitely need to be updated in the readme. I was unable to get automatic deployment to work (deploying the package did not actually install anything), but using @levmin 's instructions I managed to get manual deployment working. Thanks!

loociano, 2019-01-22:
Thank you @levmin, this instructions were extremely helpful.
I had disabled Secure Boot, installed the certificate and even enabled Win10 Test Mode without success. As soon as I disabled Signature Enforcement, the drivers were installed successfully.

NathanMAdS, 2020-05-11:
Hi how to disable Secure Boot,

boger047, 2020-05-11:
BIOS-setting-security-secure boot 
choose disable

closed 2025-11-11 by v-junyli