microsoft/Windows-driver-samples #185: DDProxy Can not proxy DNS

smallb, 2017-12-18:
I try to proxy DNS but I failed.
how to make it work?
thanks!

2812140729, 2017-12-28:
I have same issue . the outbound data can be redirected to remote endpoint , but remote-reply can't reach the application  while  the wireshark or microsoft network monitor can capture the reply data from the network adapter. I tried  debug the driver , found that the inbound traffic didn't trigger the inbound filter so the classifyfn didn't work , it's Weird

dingguanghui, 2018-01-01:
why docs it's not full? Microsoft have a mistake?

gowinder, 2018-06-12:
I found this problem too, I first think is change second DDProxyAddFilter's parameter from FWP_DIRECTION_OUTBOUND to FWP_DIRECTION_INBOUND in DDProxyRegisterFlowEstablishedCallouts .
but it still not work.

keithdg, 2022-09-07:
Did anyone figure out what the issue is with ddproxy.

The INBOUND packets are not triggering the classifyfn on Windows 10
Thanks