microsoft/Windows-driver-samples #188: Why can't find WFPSamplerProxyService module?

dingguanghui, 2018-01-01:
Hi  @pccq2002 @barrygolden.
WFPSamplerProxyService.Exe
Usage:
This is the service which listens for connections to proxy.

Â“WFPSamplerPr oxyService.Exe -iÂ“Â  will install the service.Â  While installed, the service is under t he control of the Service Control Manager (as WFPSamplerProxy).Â  The service is set to autostart.

Â“WFPSamplerPr oxyService.Exe -uÂ“ will uninstall the service.

Source:
The executable is located under ProxySvc and is comprised of the following files:

v  Framework_ProxySvc.CPP

v  Framework_ProxySvc.H

v  Framework_ProxySvc.RC

v  Framework_ProxySvc_Msg.MC

v  ThreadRoutines_Include.H

v  ThreadRoutines_TCPListener.CPP

v  ThreadRoutines_TCPListener.H

v  ThreadRoutines_UDPListener.CPP

v  ThreadRoutines_UDPListener.H

dingguanghui, 2018-01-01:
Windows-driver-samples\network\trans\WFPSampler
docs directory don't have xxxx.png. so it's very strange.
👍

gyani84, 2019-11-03:
still cant find proxysvc module. Any clue!