microsoft/Windows-driver-samples #198: input/kbfiltr/Readme.MD is wrong

timrprobocom, 2018-01-23:
There are two problems with the Readme.MD for input/kbfiltr.

First, the directions suggest replacing the INF line with ACPI\VEN_PNP&DEV_0303.  That is not the correct format for an ACPI hardware ID.  The VEN/DEV format is used by PCI, not by ACPI.  The text should either make it crystal clear that "this is just a made-up example, you'll need to substitute the actual hardware ID for your device", or else it should use the correct PS/2 identifier, ACPI\PNP0303, or perhaps use a USB\VID_xxxx&PID_xxxx example, since that's what 99.9% of users will actually have.

More importantly, the directions include the totally misleading suggestion to use "devcon install" to install the filter.  "Devcon install," despite the deceptively attractive name, cannot be used to associate an INF file with actual hardware.  What "devcon install" does is create a brand new FAKE device with the hardware ID specified, and then use the INF to associate a driver with that fake device.  Perhaps a more productive suggestion would be to use "pnputil -i -a" to install the driver package.

I would ordinarily do a pull request to submit a fix, but the philosophy here needs to be officially blessed.

themummy, 2018-03-01:
I'm having issues installing the driver using devcon as well. Were you able to successfully manually install it? If so, what's the command that you used?

timrprobocom, 2018-03-01:
Did you read my comment?  I explain why "devcon install" will not work, and suggest a better choice.  You can also "Update Driver" from Device Manager.

themummy, 2018-03-01:
Thanks, @timrprobocom. I was able to install it by right-clicking the .inf file and "Install". But I'm curious to know the exact "pnputil" command that you used.

closed 2025-11-11 by v-junyli