microsoft/Windows-driver-samples #219: kbfiltr devcon.exe fails to install driver

themummy, 2018-03-01:
**Windows 10/WDK 10/ Visual Studio 2017**

Using the the command `Devcon install kbfiltr.inf ACPI\VEN_PNP&DEV_0303` produces the following output:
```
Device node created. Install is complete when drivers are installed...
Updating drivers for ACPI\VEN_PNP&DEV_0303 from E:\GitHub\Sandbox\Windows-driver-samples\input\kbfiltr\sys\Debug\kbfiltr\kbfiltr.inf.
devcon.exe failed.
```

Furthermore, looking at the `setupapi.dev.log` under `C:\Windows\inf` it seems like it can't match the driver:
```
>>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - ACPI\VEN_PNP&DEV_0303]
>>>  Section start 2018/03/01 11:46:56.463
      cmd: "C:\Program Files (x86)\Windows Kits\10\Tools\x64\devcon.exe" update E:\GitHub\Sandbox\Windows-driver-samples\input\kbfiltr\sys\Debug\kbfiltr\kbfiltr.inf ACPI\VEN_PNP&DEV_0303
     ndv: INF path: E:\GitHub\Sandbox\Windows-driver-samples\input\kbfiltr\sys\Debug\kbfiltr\kbfiltr.inf
     ndv: Install flags: 0x00000001
     ndv: {Update Device Driver - ROOT\KEYBOARD\0000}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'E:\GitHub\Sandbox\Windows-driver-samples\input\kbfiltr\sys\Debug\kbfiltr\kbfiltr.inf'
     dvi:      {Build Driver List} 11:46:56.531
     dvi:           Searching for hardware ID(s):
     dvi:                acpi\ven_pnp&dev_0303
     dvi:      {Build Driver List - exit(0x00000000)} 11:46:56.567
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 11:46:56.586
     dvi:           Default installer: Enter 11:46:56.598
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed!
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 11:46:56.642
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\KEYBOARD\0000'.
     ndv: {Update Device Driver - ACPI\PNP0303\4&16F9A89D&0}
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'E:\GitHub\Sandbox\Windows-driver-samples\input\kbfiltr\sys\Debug\kbfiltr\kbfiltr.inf'
     dvi:      {Build Driver List} 11:46:56.702
     dvi:           Searching for hardware ID(s):
     dvi:                acpi\ven_pnp&dev_0303
     dvi:                acpi\pnp0303
     dvi:                *pnp0303
     dvi:      {Build Driver List - exit(0x00000000)} 11:46:56.740
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 11:46:56.756
     dvi:           Default installer: Enter 11:46:56.765
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed!
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 11:46:56.810
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ACPI\PNP0303\4&16F9A89D&0'.
!    ndv: No devices were updated.
<<<  Section end 2018/03/01 11:46:56.862
<<<  [Exit status: FAILURE(0x00000103)]
```

kakopappa, 2018-08-22:
i have the same problem. did you ever findout why.?

themummy, 2018-08-22:
@alevkov, do you remember what resolved this issue?

ParcelRot, 2019-06-30:
I am having the same problem.

jasonknichel, 2022-12-03:
What device IDs does your version of kbfiltr.inf match on?  The current version only matches on *PNP0BAAD, not ACPI\VEN_PNP&DEV_0303 

[Standard.NTamd64]
%kbfiltr.DeviceDesc% = kbfiltr, *PNP0BAAD

Assuming you haven't modified the INF (and assuming it hasn't been changed in the years since this issue was opened), the fact that kbfiltr.inf doesn't match on ACPI\VEN_PNP&DEV_0303 would be why it will not install on the device.

closed 2022-12-03 by jasonknichel