microsoft/Windows-driver-samples #236: Cannot compile XPSDrvSmpl

shenjeff, 2018-04-23:
Hello,

I am trying to compile the XPSDrvSmpl following the instructions. From what I can find online the Signability Test failed usually occurs when one forgets to add the install folder items to the project. However for me, I can't seem to get the dll's of the referenced project to copy over. I am using Visual Studio Community 2017 with WDK 10.0.16292.1001 to try compile this project but get the following result:

```
Inf2Cat Tool Output:
........................................
Signability test failed.

Errors:
22.9.1: xdsmplui.dll in [configplugin] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdwmark.dll in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdcolman.dll in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdbook.dll in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdnup.dll in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdscale.dll in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.

Warnings:
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntx86] of package\xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntia64] of package\xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_pre_vista in section [standard.ntamd64] of package\xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntx86.6.0] of package\xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntia64.6.0] of package\xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntamd64.6.0] of package\xdsmpl.inf
22.9.3: Missing hardware ID (cannot have just compatible ID) in line xpsdrv sample driver=install_xdsmpl_filters_vista in section [standard.ntarm64.6.0] of package\xdsmpl.inf
```

This is what my package project looks like: 

[image]

As you can see the missing DLL's are referenced but not being found.

Any help into fixing this issue?

Thanks,

Jeff

dougludlow, 2018-08-17:
Seeing the same exact error.

dougludlow, 2018-08-17:
I believe I've got a solution for this. In the [README](https://github.com/Microsoft/Windows-driver-samples/blob/c97def4e22793a07317c1c0170fd000f995765dd/print/XPSDrvSmpl/README.md) it says:

> In the SourceDisksFiles section, change the location of the DLL files you are building to =1. This indicates that there is no architecture specific directory in this driver. If you ship multiple architectures simultaneously, you will need to collate the driver INF manually.

I didn't understand this, so I didn't do anything, especially since most of the entries were already set to `=1`. However it appears that all values need to be `=1`. So this section of the `xdsmpl.inf`:

```
[SourceDisksFiles]
xdsmpl.gpd                    = 1
xdnames.gpd                   = 1
xdwmark.gpd                   = 1
xdbook.gpd                    = 1
xdcolman.gpd                  = 1
xdnup.gpd                     = 1
xdpgscl.gpd                   = 1
xdsmpl-pipelineconfig.xml     = 1
xdsmpl.ini                    = 1
xdwscRGB.icc                  = 1
xdCMYKPrinter.icc             = 1
xdwmark.dll                   = 2
xdcolman.dll                  = 2
xdbook.dll                    = 2
xdnup.dll                     = 2
xdsmplui.dll                  = 2
xdscale.dll                   = 2
```

Should end up looking like this:

```
[SourceDisksFiles]
xdsmpl.gpd                    = 1
xdnames.gpd                   = 1
xdwmark.gpd                   = 1
xdbook.gpd                    = 1
xdcolman.gpd                  = 1
xdnup.gpd                     = 1
xdpgscl.gpd                   = 1
xdsmpl-pipelineconfig.xml     = 1
xdsmpl.ini                    = 1
xdwscRGB.icc                  = 1
xdCMYKPrinter.icc             = 1
xdwmark.dll                   = 1
xdcolman.dll                  = 1
xdbook.dll                    = 1
xdnup.dll                     = 1
xdsmplui.dll                  = 1
xdscale.dll                   = 1
```

When I made that change, it built just fine.

nebula2, 2018-12-19:
Same issue here,

[image]

the SourceDisksFiles area in the inf-file is correctly set to 1.
I can't help it :(

dreamer108, 2019-03-18:
Hello! I have same issue :(
Changing values near the names of dll files does not affect anything
[image]

I get these errors on any build configuration

Additionaly, i have a strange warning:
Warning	MSB8038	Spectre mitigation is enabled but Spectre mitigated libraries are not found.  Verify that the Visual Studio Workload includes the Spectre mitigated libraries.  See https://aka.ms/Ofhn4c for more information.

Somebody please help:(

Exorcismus, 2019-05-11:
actually it compiled when you check all to 1 under [SourceDisksFiles]

Prakash19921206, 2021-08-20:
https://stackoverflow.com/questions/56085913/inf2cat-driverver-set-to-a-date-in-the-future

Eager2LearnMan, 2023-02-10:
I'm experiencing the same problem. I am using Visual Studio Community 2022 with WDK 10.0.22621.382.
I tried what was said but no result

`Inf2Cat Tool Output:
.............................
Signability test failed.

Errors:
22.9.1: xdwscrgb.icc in [colorprofiles] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdcmykprinter.icc in [colorprofiles] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl.ini in [configplugin] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl.gpd in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdnames.gpd in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdwmark.gpd in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdbook.gpd in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdcolman.gpd in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdnup.gpd in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdpgscl.gpd in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: xdsmpl-pipelineconfig.xml in [xpsdrvsample] of package\xdsmpl.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.

Warnings:
None`