microsoft/Windows-driver-samples #241: WFPSample error in running

rakeshlacework, 2018-05-01:
Getting an error when I run WFPSample.exe - 

[image]

My environment -  
* MSVC 2017
* WDK 10.0.17134.0 
* SDK 10.0.16299.91

Can anybody help with some pointers

isdqwq, 2018-05-09:
Function FwpmProviderContextAdd3 location at file  FWPUCLNT.DLL
For windows Server 2012 R2 
please check the function had export on file FWPUCLNT.DLL

Brian-E, 2018-10-16:
For 2012 R2 you have to build against the 8.1 SDK. I'm pulling my hair out trying to do that right now so if anyone has had any success please let me know.

ronglianglei, 2021-05-07:
Is Windows Server 2012 R2 still supporting ? I m using visual studio 2019 Professional. I used WDK 10 to build.  WDK 8.1 is not available to download. https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads . I found the WFPSamplerService has same problem.