microsoft/Windows-driver-samples #265: SYSVAD does not pass Certification Process Failures...

vburel2018, 2018-07-24:


vburel2018, 2018-07-25:
The following test could become successful on second pass, after having performed all other tests.
-Audio Codec – AC3 Test – Certification – Desktop
-Audio Codec – Class Driver Audio Logo Test – Certification – Desktop

The test below could also be successful for another reason 
-Audio Codec – Audio Logo Test – Glitch – Desktop

Pending on installation process ? Different default audio format ? Or just a timing chance ?

vburel2018, 2018-07-25:
The result of the 3 MFCaptureEngine API tests, seems to belong to the other drivers installed on the test machine... According test log, HLK performs test on all audio drivers present on the system (not only on SYSVAD): If i just let the Motherboard Realtek High Definition Audio driver, then these tests become successful: 
-Audio Codec – MFCaptureEngine API Test – Capture AMR-NB audio to a file from the default capture pin
-Audio Codec – MFCaptureEngine API Test – Capture PCM audio samples from the default capture pin
-Audio Codec – MFCaptureEngine API Test – Capture PCM audio to a wav file from the default capture pin

**it remains the Audio Codec – KS Position Test**...

mvaneerde, 2018-07-31:
Please file a bug on http://aka.ms/Collaborate for each failing test, and attach to each bug an .hlkx with the failing test result.

mvaneerde, 2018-08-01:
[Collaborate docs > Feedback > Submit and Manage](https://docs.microsoft.com/en-us/collaborate/feedback-items-manage)

vburel2018, 2018-08-14:
Re-performing Certification tests on SYSVAD from the beginning, on fresh WIN10 installation and using devcon correctly is producing the following errors:

**-Audio Codec – AC3 Test – Certification – Desktop**
- KS Pin Tests\KSCreatePin Multiple (SYSVAD wave HDMI)
- KS Pin Tests\KSCreatePin Multiple (SYSVAD wave SPDIF)

**-Audio Codec – Class Driver Audio Logo Test – Certification – Desktop**
- Glitch Testing\verify n-stream capture is glitch free (Internal Microphone Array Rear)
- Glitch Testing\verify n-stream stressed capture is glitch free (Internal Microphone Array Front/Rear)
- Glitch Testing\verify n-stream stressed capture is glitch free (External Microphone Headphone)
- Glitch Testing\verify n-stream render is glitch free (SPDIF Interface)
- Glitch Testing\verify n-stream stressed render is glitch free (SPDIF Interface)
- Glitch Testing\verify n-stream stressed render is glitch free (Speakers)
- Glitch Testing\verify n-stream render is glitch free (Headphone)
- Glitch Testing\verify n-stream stressed render is glitch free (Headphone)
- Glitch Testing\verify n-stream render is glitch free (SinkDescription Sample)

**-Audio Codec – Communications Audio Fidelity Test (Manual) – Desktop**
- Cannot find Reference device with Loudspeaker and Microphone… (this test passed successfully on our previous certification test sessions)…

**-Audio Codec – MFCaptureEngine API Test – Capture PCM audio samples from the default capture pin**
- no error notified in wtl file…

The second pass changed nothing (re-performing failed tests makes no change) except we got different error list for:

**-Audio Codec – Class Driver Audio Logo Test – Certification – Desktop**
- Glitch Testing\verify n-stream capture is glitch free (Internal Microphone Array Front)
- Glitch Testing\verify n-stream capture is glitch free (Internal Microphone Array Front/Rear)
- Glitch Testing\verify n-stream capture is glitch free (External Microphone Headphone)
- Glitch Testing\verify n-stream stressed capture is glitch free (External Microphone Headphone)
- Glitch Testing\verify n-stream render is glitch free (SinkDescription Sample)
- Glitch Testing\verify n-stream stressed render is glitch free (SinkDescription Sample)
- Glitch Testing\verify n-stream render is glitch free (Speakers)
- Glitch Testing\verify n-stream render is glitch free (Headphone)
- Glitch Testing\verify n-stream stressed render is glitch free (Headphone)
- Glitch Testing\verify n-stream render is glitch free (SPDIF Interface)
- Glitch Testing\verify n-stream stressed render is glitch free (SPDIF Interface)

Does someone got the same experience ?

darugeri, 2020-09-11:
Hello,

For the tests you are mentioning I have the following remarks:
- Audio Codec – AC3 Test – Certification – Desktop: This test is passing with current SysVAD,
- Audio Codec – Class Driver Audio Logo Test – Certification – Desktop:      Glitch testing was dropped and is no longer included in HLK,
- Audio Codec – MFCaptureEngine API Test – Capture PCM audio samples from the default capture pin: I cannot find this test in the current HLK. Can you confirm the name and if it is still included in HLK?
- Audio Codec – Communications Audio Fidelity Test (Manual) – Desktop: This test is not included in the audio playlist and it is not applicable on SysVAD since it depends on real hardware.