microsoft/Windows-driver-samples #268: SYSVAD manual installation fails on Target PC with devcon.exe

vburel2018, 2018-07-31:
we recompiled a SYSVAD Audio Tablet Sample Driver with Latest Source Code 7/23/2018 version 10.41.14.806 And we tried to install it on our test machine: WIN10 PRO Build 17134.RS4 Release 180410-1804 to perform certification test.

we use the procedure given in the readme file:
C:\tools\devcon install tabletaudiosample.inf *ROOT\sysvad_TabletAudioSample 

but it does not work... the procedure is declared FAILED and the device manager confirm that SYSVAD is not correctly installed as audio device...

REM : this 2 years old issue is also still opened:
https://github.com/Microsoft/Windows-driver-samples/issues/57

mvaneerde, 2018-07-31:
Thanks for filing the issue. `C:\Windows\inf\setupapi.dev.log` should have more detail on why `devcon.exe install ...` failed. Can you share it? The most relevant lines usually begin with `!!!`

vburel2018, 2018-07-31:
```
>>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - *ROOT\sysvad_TabletAudioSample]
>>>  Section start 2018/07/31 17:53:42.444
      cmd: "C:\tools\devcon.exe" install tabletaudiosample.inf *ROOT\sysvad_TabletAudioSample
     ndv: INF path: C:\VBTest\sysvad\tabletaudiosample.inf
     ndv: Install flags: 0x00000001
     ndv: {Update Device Driver - ROOT\MEDIA\0000}
     dvi:      Set selected driver complete.
     ndv:      Search options: 0x00000080
     ndv:      Searching single INF 'C:\VBTest\sysvad\tabletaudiosample.inf'
     dvi:      {Build Driver List} 17:53:42.540
     dvi:           Searching for hardware ID(s):
     dvi:                *root\sysvad_tabletaudiosample
     dvi:           Processing a single INF: 'c:\vbtest\sysvad\tabletaudiosample.inf'
     inf:           Opened INF: 'c:\vbtest\sysvad\tabletaudiosample.inf' ([strings])
     dvi:           Enumerating INFs from path list 'C:\Windows\INF'
     inf:           Searched 0 potential matches in published INF directory
     dvi:      {Build Driver List - exit(0x00000000)} 17:53:42.587
!    ndv:      No matching drivers found in single INF
     dvi:      {DIF_SELECTBESTCOMPATDRV} 17:53:42.589
     dvi:           Using relative path for module 'C:\Windows\system32\mmci.dll'.
     dvi:           Using exported function 'MediaClassInstaller' in module 'C:\Windows\system32\mmci.dll'.
     dvi:           Class installer == mmci.dll,MediaClassInstaller
     dvi:           No CoInstallers found
     dvi:           Class installer: Enter 17:53:42.604
     dvi:           Class installer: Exit
     dvi:           Default installer: Enter 17:53:42.620
     dvi:                {Select Best Driver}
!    dvi:                     Selecting driver failed(0xe0000228)
     dvi:                {Select Best Driver - exit(0xe0000228)}
!    dvi:           Default installer: failed
!    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
     dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 17:53:42.654
!    ndv:      Unable to select best compatible driver. Error = 0xe0000228
     ndv:      No drivers found for device.
     ndv: {Update Device Driver - exit(00000103)}
!    ndv: No better matching drivers found for device 'ROOT\MEDIA\0000'.
!    ndv: No devices were updated.
     dvi: {DIF_DESTROYPRIVATEDATA} 17:53:42.670
     dvi:      Class installer: Enter 17:53:42.670
     dvi:      Class installer: Exit
     dvi:      Default installer: Enter 17:53:42.686
     dvi:      Default installer: Exit
     dvi: {DIF_DESTROYPRIVATEDATA - exit(0xe000020e)} 17:53:42.686
<<<  Section end 2018/07/31 17:53:42.701
<<<  [Exit status: FAILURE(0x00000103)]
```

mvaneerde, 2018-07-31:
Thanks. Can you look in `C:\VBTest\sysvad\tabletaudiosample.inf` for a line that looks either like `[SYSVAD.NT$ARCH$]` or like `[SYSVAD.NTamd64]` or similar?

If you find `[SYSVAD.NT$ARCH$]` then you're running devcon.exe against the pre-stamped .inf (which should, more properly, have the .inx extension), which won't work. Visual Studio should automatically stamp the .inf as part of the build process, and put the stamped .inf in the build output.

You can also run [stampinf.exe](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/stampinf-command-options) directly; the relevant argument is `-a [architecture]`

mvaneerde, 2018-07-31:
Yup - you can't pass that to devcon.exe directly, it has to be built first.

When you build the driver, you specify the architecture (x86, x86-64, etc.); the .cpp files are built into .sys files with the right machine code for the architecture you specify, and the .inf file (which should really be a .inx) has all of the `$ARCH$` wildcards replaced with the right architecture-specific string.

vburel2018, 2018-07-31:
of course i don't use this file directly , but the generated file is coming from this... with the project given here....

mvaneerde, 2018-07-31:
OK, so you're using the stamped .inf. Good.

> !    ndv:      No matching drivers found in single INF

This is definitely the problem... _scratches chin_

The .inf has this:
> %SYSVAD_SA.DeviceDesc%=SYSVAD_SA, Root\sysvad_TabletAudioSample

And the readme has this:
> Use Ctrl+F to find the [MicrosoftDS] section. Note that there is a comma-separated element at the end of the row. The element after the comma shows the hardware ID. So for this sample, the hardware ID is *ROOT\sysvad_TabletAudioSample.

I think the readme is wrong (well, the steps are right, but the conclusion is wrong). The hardware ID is `Root\sysvad_TabletAudioSample`, not `*ROOT\sysvad_TabletAudioSample`. Try removing the * from the devcon.exe command line.

vburel2018, 2018-07-31:
ha, ok, now it works! the right command line is 
C:\tools\devcon install tabletaudiosample.inf ROOT\sysvad_TabletAudioSample

mvaneerde, 2018-07-31:
Thanks, I've filed a bug internally to update the doc.

DanijelDomazet, 2020-12-07:
`C:\\tools\\devcon install ComponentizedAudioSample.inf Root\Sysvad_ComponentizedAudioSample`
Is that the right command line?

mvaneerde, 2020-12-07:
@DanijelDomazet yes, see the [README](https://github.com/microsoft/Windows-driver-samples/blob/master/audio/sysvad/README.md) for the correct command line.

DanijelDomazet, 2020-12-16:
OK, that gets me half way done. 
I have installed: 
1. SUCCESS: `devcon install ComponentizedAudioSample.inf Root\Sysvad_ComponentizedAudioSample`
2. SUCCESS: right click on `ComponentizedAudioSampleExtension.inf` and `install`
but the step 3 fails:
3. FAIL: right click on `ComponentizedApoSample.inf` and `install`

Do you know what the error `dvs: {DrvSetupInstallDriver - exit(00000103)}` means? 
The log is attached.

[apo_install_error_log.txt](https://github.com/microsoft/Windows-driver-samples/files/5702748/apo_install_error_log.txt)

DanijelDomazet, 2020-12-16:
Is there a command line I could try instead of right-click and `install`?