microsoft/Windows-driver-samples #269: msnmntr的小错误?

kouzhudong, 2018-08-08:
in Windows-driver-samples/network/trans/msnmntr/sys/notify.c

i found:

      if ((_strnicmp((const char*)stream, "POST", streamLength) == 0)
          || (_strnicmp((const char*)stream, "GET", streamLength) == 0))

and 

      if (_strnicmp((const char*)stream, "HTTP", streamLength) == 0)

never success

----------------------------------------------------------------------------

解决办法,换为不区分大小写的字符串搜索函数.
或者把_strnicmp的第三个参数改为第二个字符串参数的长度,但是得断言这个长度小于等于streamLength.

这么个简单的问题,不知大家测试的时候发现没?

kruglinski, 2018-08-09:
Hi correy!

I am your ex-coworker jjwang, glad to see you here!

_strnicmp it is case insensitive compare

I haven't trace these code yet, but by review it I think it means that
to match "POST", "GET", "HTTP" from stream with case insensitive and
don't cross the buffer boundary.

you said that "never success" mean the stream doesn't match these
words, doesn't mean that it is a bug cause "never success"

here is some tests for _strnicmp:

int main(int argc, char **argv)
{
    char buf1[100]="post\0";
    char buf2[100]="get\0";
    char buf3[100]="http\0";

    cout.setf(ios_base::boolalpha);

    auto match = [](char *stream, size_t streamLength)
    {
        cout<<"  POST: "<<(_strnicmp(stream, "POST", streamLength)==0)<<endl;
        cout<<"  GET: "<<(_strnicmp(stream, "GET", streamLength)==0)<<endl;
        cout<<"  HTTP: "<<(_strnicmp(stream, "HTTP", streamLength)==0)<<endl;
        cout<<endl;
    };

    cout<<"test post:"<<endl;
    match(buf1, 100);

    cout<<"test get:"<<endl;
    match(buf2, 100);

    cout<<"test http:"<<endl;
    match(buf3, 100);

    return 0;
}

and it outputs:

test post:
  POST: true
  GET: false
  HTTP: false

test get:
  POST: false
  GET: true
  HTTP: false

test http:
  POST: false
  GET: false
  HTTP: true

kouzhudong, 2018-08-09:
这种专门测试是可以的.

正常情况下是很少发生的.

kruglinski, 2018-08-09:
stream中关键词后面没有\0终止？那样的话是会有问题的！

kouzhudong, 2018-08-09:
_strnicmp的第三个参数限制了比较的个数.

closed 2025-11-11 by v-junyli