microsoft/Windows-driver-samples #270: SysVAD - Change "External Microphone Headphone" in "Microphone"

andeb91, 2018-08-08:
Hi to all. I don't know if this is the right place, but I try...

I'm working with SysVAD example and I would like to know if there is a way to change one of the input sources, like "External Microphone Headphone", in "Microphone". I understood that is the name of the virtual device category but I can't find a way to switch the category. I need this because in my application is better that users see Microphone instead everything else (I don't want to use Microphone Arrays too).

Thanks in advance to everybody who wants help me. 

Andrea

girishpattabiraman, 2018-08-14:
Hi Andrea,

Can you give a bit more details about what you want to change. I can try it out.

Changing pin categories is easy. There are mic*toptable.h files which you can modify.

Regards,
Girish

v-junyli, 2025-11-11:
https://github.com/microsoft/Windows-driver-samples/issues/226

closed 2025-11-11 by v-junyli