microsoft/Windows-driver-samples #281: Building any kernel mode driver project in Debug configuration fails with VS2017

assarbad, 2018-08-31:
Example:

```
1>Building 'tracedrv' with toolset 'WindowsKernelModeDriver10.0' and the 'Desktop' target platform.
1>tracedrv.c
1>tracedrv.obj : error LNK2019: unresolved external symbol @__CheckForDebuggerJustMyCode@4 referenced in function _DriverEntry@8
1>Debug\tracedrv.sys : fatal error LNK1120: 1 unresolved externals
1>Done building project "tracedrv.vcxproj" -- FAILED.
```

I am reporting this here for two reasons.

1. [a corresponding post](https://developercommunity.visualstudio.com/content/problem/302014/dirver-build-debugmode-checkfordebuggerjustmycode.html) in the VS developer community got closed with status **Not a Bug**
2. the only other remedy then is to fix each and every project [as per this routine](https://www.osr.com/blog/2018/08/30/baaack-just-code-part-ii-breaks-driver-debug-builds/) or setting `<SupportJustMyCode>false</SupportJustMyCode>` inside the `<ClCompiler>` elements of the `.vcxproj` files

In my humble opinion this would be a defect in Visual Studio, since the WDK depends on the major version of Visual Studio and not on a particular minor version. So since the already released WDK can't be fixed, the VS team needs to tread carefully. As far as I know MSBuild does support conditions, so there's no good reason not to make the default setting dependent on the fact that the project is linking the (user mode) C runtime, for example. But that's just me. Or, [as Peter Viscarola of ORS puts it](https://www.osr.com/blog/2018/08/16/new-vs15-8-feature-breaks-wdk-1803-workaround/) (emphasis his):

> Seeing this problem in a **released** version of Visual Studio (remember, we’re not talking about a preview release here) does beg the question:  **Doesn’t anybody test this stuff?**  Are we, the Windows third-party driver development community, the first people to use new version of these tools to build drivers?  If so, dude… that’s scary.  Really scary.

I agree, it is really scary, because it looks like no one in the VS team bothers to build a driver then, or they are using some newer internal version of the WDK to which the public has no access, and so don't see the defect.

But maybe you good folks at MSFT can figure out among each other whether that's a defect ("bug") or not -- hint: yes it is -- and where the defect is rooted.

Thanks and best regards,

Oliver

PS: [related](https://stackoverflow.com/q/51853537)

NeoAdonis, 2018-09-04:
Apologies for the problems you're currently having. We're aware of this issue and we (the WDK team) are working closely with Visual Studio to both fix this and to have a concise plan that allows us to prevent in the future that something like this happens again.

closed 2018-09-04 by NeoAdonis

serup, 2018-09-11:
This issue should only occur for VS2017 versions lower that 15.8

ArbelDuff, 2018-09-12:
I use VS 15.8.3, having the same issue.

assarbad, 2018-09-14:
@serup unfortunately that's not the case. Just tried 15.8.2 (prior to the update) and 15.8.4 right after the update. Both have the issue.