microsoft/Windows-driver-samples #297: Not Able to install KMDF Echo driver on Windows7 x64

ankurbindal94, 2018-10-12:
I am trying to deploy KMDF Echo driver on a test Windows 7 machine to get started with WDK 8.1

While building i selected for a test signature, and installed the certificate generated as a trusted root certificate onto the test system(also checked the stampInf settings - it has InfArch set to amd64).
After building it successfully, i copied the echo.sys,echo.inf,kmdfsamples.cat and WdfCoinstaller01011.dll to the test system. Copied devcon.exe to the same folder and a shortcut to cmd
Ran cmd as administrator and executed devcon command - "devcon install echo.inf root\ECHO" and i get the following error in setupapi.dev.log

     dvi: {DIF_SELECTBESTCOMPATDRV} 13:15:38.638
     dvi:      No class installer for 'Unknown driver software package'
     dvi:      No CoInstallers found
     dvi:      Default installer: Enter 13:15:38.640
     dvi:           {Select Best Driver}
!    dvi:                Selecting driver failed(0xe0000228)
     dvi:           {Select Best Driver - exit(0xe0000228)}
!    dvi:      Default installer: failed!
!    dvi:      Error 0xe0000228: There are no compatible drivers for this device.
     dvi: {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 13:15:38.643
<<<  Section end 2018/10/12 13:15:38.644
<<<  [Exit status: SUCCESS]

i even tried this after running 'bcdedit /set nointegritychecks ON' and still got the same error

Another thing i tried was to add the driver using devcon dp_add command which gave the following log - 
!    sig:                     Verifying file against specific Authenticode(tm) catalog failed! (0x800b0109)
!    sig:                     Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:                {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 13:08:55.283
!    sto:                Certificate associated to driver package is not present in Trusted Root Certificates, but user wants to install anyway.

But the Test Certificate i used is indeed present in the Trusted Root Certificate List.
No Solution i came across online seems to work, any would be appreciated

Thanks

CodeAbyss, 2018-10-12:
Try this once, 
[bcdedit /set TESTSIGNING ON](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/the-testsigning-boot-configuration-option)

CodeAbyss, 2018-10-12:
You have reboot'ed the machine, right? Because the bcdedit will take effect on reboot. Also, I think the cat file needs to be renamed same as sys file, i.e., echo.sys. Also, you can check event viewer for detailed error.

ankurbindal94, 2018-10-12:
> You have reboot'ed the machine, right? Because the bcdedit will take effect on reboot. Also, I think the cat file needs to be renamed same as sys file, i.e., echo.sys.
> 
> Also, you can check event viewer for detailed error. Try, Event Viewer -> Applications and services Logs -> Microsoft -> Windows -> CAPI2 -> Operational [Right Click -> Enable log]. This will give a better insight into the problem.
> 
> Can you share the cmd you are using to create the Test Cert? You need not install the test cert in testmode machine.

Yes i did reboot the machine after doing BCD settings, faced the same issue after that.

Tried the event viewer for a detailed log like you suggested, but is shows similar error to what i got with devcon dp_add- 
StepError        [  stepID]  3       [  stepName]  TRUSTERROR_STEP_SIP          - Result The form specified for  the subject is not one supported or known by the specified trust  provider.        [  value]  800B0003 

I haven't used any command in particular, simply selected this under Driver Signing in project properties while building driver package in VS2013

CodeAbyss, 2018-10-12:
If you want, you can try this.
1. Turn off the Driver signing under VS2013.
2. Use the cmd to generate Test Cert "**makecert.exe -a SHA256 -r -pe -ss my -n "CN=Test" -eku 1.3.6.1.5.5.7.3.3 Test.cer**"
3. Use the cmd to sign the driver "**signtool.exe sign /fd SHA256 /a /v /ph /sha1 <SHA1_Thumbprint> Echo.sys**" (_Use /ph if CodeIntegrity is enabled for the binary. IMHO, you should enable the CI for your driver_)

ankurbindal94, 2018-10-15:
> If you want, you can try this.
> 
> 1. Turn off the Driver signing under VS2013.
> 2. Use the cmd to generate Test Cert "**makecert.exe -a SHA256 -r -pe -ss my -n "CN=Test" -eku 1.3.6.1.5.5.7.3.3 Test.cer**"
> 3. Use the cmd to sign the driver "**signtool.exe sign /fd SHA256 /a /v /ph /sha1 <SHA1_Thumbprint> Echo.sys**"
> 
> * (_Use /ph if CodeIntegrity is enabled for the binary. IMHO, you should enable the CI for your driver_)
> * (_Sha1 Thumbprint should be without spaces & can be found by double-clicking the certificate file -> Details -> ThumbPrint_) (Be very cautious while copying the Thumbprint, better to read and type.)

I Tried this, but i get an error - "The System cannot find the file specified" when i run signtool.exe
Also i am not aware how to ensure if Code Integrity is enabled for the driver or not, can you tell where i can find this?