microsoft/Windows-driver-samples #337: Can't install WIA driver on Windows 10 with VS 2017 Community

matteoacrossi, 2019-01-22:
I tried to build the WIA driver sample with VS 2017 community. I'm able to build successfully, but then the `copywia.cmd` script fails due to release paths being different. 

I copied the files manually, but when I add the device using the Add Device from the Scanners and cameras window > Have Disk... I can choose any of the _Wia sample microdriver_, _Production Scanner Test Driver_ or _Extended WIA Monster device_, I get "This driver is not digitally signed" warning, and then nothing happens if i click Next.

How do I build the drivers so that they have a digital signature on the same machine (at least for testing)?

sanbrother, 2019-04-21:
@matteoacrossi A signed category file (.cat) is needed under x64 OS.
[https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation)  

For test purpose, it is easy to install under x86 OS.

Alexel37, 2020-03-17:
Hello,
I have build the wia project and installed 3 drivers for virtual 32 bit machine (windows 8)
but any of 3 scanners are not working could you provide some information what to do next?
[image]
[image]