microsoft/Windows-driver-samples #360: avssamp installs but does not work

timrprobocom, 2019-03-28:
The AVStream avssamp sample builds and installs correctly, but the filter never appears in the DirectShow enumeration.  Graphedt and AMCap do not find it.  Ksstudio sees it, but comes up with the wrong HardwareID and no friendly name.  I suspect a registry problem, but I haven't been able to find it.

Also, the instructions should probably recommend "pnputil -a" instead of "devcon install".  Much less error prone.

timrprobocom, 2024-01-17:
I was actually reporting this for someone else.  I revamp the sample INFs before I try to use them.

shiqifeng2000, 2024-01-18:
> INFs

so the problem was with the inf file? may I know where I should fix to get it working?

walker-WSH, 2025-05-20:
hi @timrprobocom  ,  can you share your solution ?  thanks