microsoft/Windows-driver-samples #362: Retrieving PrintTicket for shared XPS driver failes (returns PrintTicket of local computer)

wstaelens, 2019-04-12:
When retrieving the PrintTicket from a shared XPS v4 driver, you get the PrintTicket of the local computer and not of the computer who printed to the shared driver.

The IPrintPipelinePropertyBag contains the incorrect information. 
The user can be found when parsing the  xps datastream itself.

Either this is a bug, or documentation should be improved as it is not clear then what we are doing wrong.

```
__override STDMETHODIMP StreamFilter::InitializeFilter(
    _In_    IInterFilterCommunicator         *pIFilterCommunicator,
    _In_    IPrintPipelinePropertyBag        *pIPropertyBag,
    _In_    IPrintPipelineManagerControl     *pIPipelineControl
    ){
…
m_JobInfo = new JobInformation(pIPropertyBag);
}

void JobInformation::SavePrintTicket(CString outputFolder)
{
SafeVariant varUserPrintTicket;
VariantInit(&varUserPrintTicket);
THROW_ON_FAILED_HRESULT(m_pIPropertyBag->GetProperty(XPS_FP_USER_PRINT_TICKET, &varUserPrintTicket));
IUnknown_t pUnk = varUserPrintTicket.punkVal;

IPrintReadStreamFactory_t pStreamFactory;
THROW_ON_FAILED_HRESULT(pUnk.QueryInterface(&pStreamFactory));

//
// Get the default user Print Ticket stream
// and wrap it in an IStream
//
IPrintReadStream_t pUserPrintTicketStream;
THROW_ON_FAILED_HRESULT(pStreamFactory->GetStream(&pUserPrintTicketStream));

DumpPrintTicket(pUserPrintTicketStream, outputFolder);
}
```

elvis-sun, 2022-11-21:
@wstaelens 
hi, Could you tell me how to save xps document to file?

wstaelens, 2022-11-21:
@elvis-sun what do you mean? I think the provided info is clear enough.

elvis-sun, 2022-11-21:
@wstaelens 
I want to get xps file data from xps driver. 
I built xpsdrvsmple and add printer using the driver. I print txt file and get xps file.  Now I want to get xps file date from xps drvier, but I don't know how to get it from driver.
as for me, I am confused about printer driver.
Please give me some advice or some link about xps printer.

elvis-sun, 2022-11-21:
> @wstaelens I want to get xps file data from xps driver. I built xpsdrvsmple and add printer using the driver. I print txt file and get xps file. Now I want to get xps file date from xps drvier, but [...]

@wstaelens 
???

closed 2025-11-11 by v-junyli

v-junyli, 2025-11-12:
@wstaelens reopened. Thank you for the feedback that this issue is still relevant.

@microsoft/core-print can you please take a look if your team owns this sample?