microsoft/Windows-driver-samples #366: Multiple examples indicate "[DefaultInstall]-based INF cannot be processed as Primitive"

dovholuknf, 2019-04-30:
I am new to windows driver development and I have been trying to learn my way using the samples in the network folder.  Yesterday I made the "mistake" of upgrading visual studio and now the drivers I am using are reporting 

```C:\git\external\windows-driver-samples\network\trans\inspect\sys\inspect.inf : error 1420: [DefaultInstall]-based INF cannot be processed as Primitive.```
```
1>C:\git\external\Windows-driver-samples\network\trans\ddproxy\sys\ddproxy.inf : error 1420: [DefaultInstall]-based INF cannot be processed as Primitive.```

This page discusses the issue: https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-primitive-driver however the samples are not updated at this time.

mutterbucket, 2019-08-07:
A solution that worked for me is to right click the driver project, and in the Driver Settings section change the Target Platform from Universal to Desktop.

dovholuk, 2019-08-08:
> A solution that worked for me is to right click the driver project, and in the Driver Settings section change the Target Platform from Universal to Desktop.

This allows the project to build for me - but it won't actually deploy to the target machine. For me - the errors are just turned into warnings.

After you made that change - are you able to deploy the app? I get:
`Extracting service name from inf "C:\git\external\windows-driver-samples\network\trans\inspect\sys\x64\Debug\inspect\inspect.inf" Failed:`

dovholuknf, 2019-08-08:
A solution that worked for me is to right click the driver project, and in the Driver Settings section change the Target Platform from Universal to Desktop.

This allows the project to build for me - but it won't actually deploy to the target machine. For me - the errors are just turned into warnings.

After you made that change - are you able to deploy the app? I get:
Extracting service name from inf "C:\git\external\windows-driver-samples\network\trans\inspect\sys\x64\Debug\inspect\inspect.inf" Failed:

mutterbucket, 2019-08-15:
I can run them on a VMWare test machine with a serial connection to WinDbg and driver signature enforcement disabled, that has been the most reliable setup I can establish right now.

estrandb, 2020-05-03:
https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-primitive-driver#legacy-compatibility

I was getting this error as well. As described in the above documentation, I decorated the DefaultInstall and DefaultUninstall attributes with .NTx86 and added LegacyUninstall = 1 to the DefaultUninstall.NTx86 section. It builds now. I don't know if it works yet though ;)

rakendrathapa, 2022-06-17:
This behavior introduced with Windows 10 version 1903 and later:
Look here: https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-primitive-driver

msojocs, 2024-04-25:
```

;;
;; Default install sections
;;

[DefaultInstall.NTamd64]
OptionDesc          = %SimRepServiceDesc%
CopyFiles           = SimRep.DriverFiles

[DefaultInstall.NTamd64.Services]
AddService          = %SimRepServiceName%,,SimRep.Service

;;
;; Default uninstall sections
;;

[DefaultUninstall.NTamd64]
LegacyUninstall=1
DelFiles   = SimRep.DriverFiles
DelReg     = SimRep.DelRegistry

[DefaultUninstall.NTamd64.Services]
DelService = SimRep,0x204

```
[image]

MrRhuezzler, 2024-05-28:
> A solution that worked for me is to right click the driver project, and in the Driver Settings section change the Target Platform from Universal to Desktop.
>
> This allows the project to build for me [...]

I am trying to kernel debug a driver, I am running into the same issue. Although I can ignore the errors and I am able to kernel debug but I need to install the driver manually in the test system after attaching the kernel debugger from visual studio.

It would be nice if someone has solved this.

closed 2025-12-17 by v-junyli