microsoft/Windows-driver-samples #373: can't install drive : cannot install printer %1 is not a valid win32 application

Exorcismus, 2019-05-11:
I managed to build XPS sample project but it keeps poping this error when I try to the add printer from wizard

[image]

those are the output folder content

[image]

what could be the issue ?

vicnroll, 2019-12-31:
I have the same problem, I built the driver but I can't install with printer wizard in Windows 10. Someone found a solution?

golddragon11, 2022-08-08:
Encountered the same problem, managed to solve it by changing the target platform of the project.

Turned out that I was trying to install the driver on a **x64** machine when it was instead compiled for **x86**.
Simply changing the target platform to **x64** worked for me.